YosysHQ/prjtrellis #20: Unsupported bitstream command 0x02 (maybe related to bitstream compression)

tinyfpga, 2018-07-22:
I'm running the `or_to_and.py` example and it fails on my bitstream with the following output:

```
bitstream size: 2245640 bits
bitstream: reset crc
bitstream: device ID: 0x41113043
Traceback (most recent call last):
  File "or_to_and.py", line 10, in <module>
    chip = bs.deserialise_chip()
ValueError: Bitstream Parse Error: unsupported command 0x02 [at 0x172]
```

This is a compressed bitstream, so I suspect this command is somehow related to compression.

You can find the entire diamond project used to generate this bitstream in this zip file: [diamond_template.zip](https://github.com/SymbiFlow/prjtrellis/files/2217443/diamond_template.zip)

daveshah1, 2018-07-22:
Project Trellis does not support compressed bitstreams. You will need to work with uncompressed bitstreams for the time being.

tinyfpga, 2018-07-25:
Since this is not a bug, but a known missing feature, I changed the title to reflect this. If anyone wants to start learning about the ECP5 bitstreams, this would be a good way to start.

tinyfpga, 2018-07-25:
bitstream compression format appears to be documented in Appendix B of this document: http://www.latticesemi.com/~/media/LatticeSemi/Documents/ApplicationNotes/EH/TN1260.pdf

tpwrules, 2019-02-24:
The document seems to reference the nonexistent "Table 103" for the documentation of compressed bits within a frame. Have you guys seen or reverse engineered this table before? It looks necessary to actually compress the data.

AndresNavarro82, 2019-03-05:
Check out this pull rquest for an explanation of the compression used and working decompression code: https://github.com/SymbiFlow/prjtrellis/pull/74
The compression should be pretty easy now.  I hope to get it done by the weekend.

emard, 2019-10-14:
Please do if possible!
compression would be good thing to kave it would save flash
space for multiboots and of course shorter bitstreams upload faster

emard, 2019-11-08:
IT works!
I tested it on ULX3S 12-25-85F, fantastic files are small and upload fast!

daveshah1, 2019-11-08:
(implemented in dc10b8a)

closed 2019-11-08 by daveshah1