YosysHQ/prjtrellis #22: Add support for advanced bitstream functionality

daveshah1, 2018-07-26:
As well as compression as mentioned in #20, it would be nice to see:

   - [ ] Dual-boot/multi-boot support
   - [ ] Dual/Quad IO flash support
   - [ ] Configuration frequency selection

olegendo, 2019-07-17:
Those things like dual/multi boot, DSPI/QSPI and config frequency have been already added, haven't they?  At least I can see some options for that in "ecppack --help"

When configuring an ECP5 in SPI slave mode (SSPI), by default the SSPI is turned off and becomes normal user IOs.  In the Lattice tool there seems to be an option Diamond > Spreadsheet
View > Global Preferences > SLAVE_SPI_POR (ENABLE).  This allows keeping the SSPI port open even after device configuration for further configuration related commands.

I guess this must be stored somewhere in the bitstream, too?  But it's not supported by trellis yet?

Mecrisp, 2022-11-08:
For ECP5, I miss an option to ecppack to keep the SPI flash active after loading of the bitstream in order to access additional data files from user logic, similiar to "icepack -s" (disable final deep-sleep SPI flash command after bitstream is loaded).

gatecat, 2022-11-08:
Does the ECP5 hardware put the flash into deep sleep? I'm not sure that it does, nor that such an option is possible in Diamond.

Mecrisp, 2022-11-08:
Thank you for the hint. I was porting code from iCE40 to ECP5 and wasn't aware that the ECP5 isn't putting flash in deep sleep after loading bitstream, which I can confirm by experiment now. Everything fine!