YosysHQ/prjtrellis #34: Create a tool like icepll for the ECP5 PLLs

daveshah1, 2018-10-21:
Although the actual config bits of the PLLs are known, how they should be set for given input and output frequencies isn't yet worked out. It would be nice to work this out and make an open source config tool similar to icepll.

C-Elegans, 2018-12-30:
I was unable to find good information on how the PLL dividers work in the ECP5 PLL Design and Usage Guide, so I ran several frequencies through Lattice's clarity designer, and was surprised with what I found:

Input | Output | DIVrefclk | DIVfeedback | DIVoutput | fvco
-- | -- | -- | -- | -- | --
12 | 48 | 1 | 4 | 12 | 576
12 | 60 | 1 | 5 | 10 | 600
20 | 30 | 2 | 3 | 20 | 600
45 | 30 | 3 | 2 | 20 | 600
100 | 400 | 1 | 4 | 1 | 400
200 | 400 | 1 | 2 | 2 | 800
50 | 400 | 1 | 8 | 2 | 800
70 | 40 | 7 | 4 | 15 | 600
12 | 36 | 1 | 3 | 18 | 648
12 | 96 | 1 | 8 | 6 | 576
90 | 40 | 9 | 4 | 15 | 600
90 | 50 | 9 | 5 | 13 | 650
43 | 86 | 1 | 2 | 7 | 602

It appears that the vco frequency is given by f_vco = f_in/ DIVrefclk * DIVfeedback * DIVoutput, and that the output frequency is given by f_out = f_vco/DIVoutput. It seems crazy to me that it would actually work this way, can anyone confirm that the PLL configuration from Clarity actually gives the correct frequency?

alibabashack, 2018-12-30:
@C-Elegans Why exactly does it seem crazy to you? Looking at Figure 23 in TN1263 I would think you are on track.

C-Elegans, 2018-12-30:
Well, every other PLL I've seen in this sort of application has been pretty similar, with an input divider, a single divider for the feedback, and an output divider. Doing it this way seems like it would reduce the frequency resolution of the PLL, and empirically it seems like it does. Perhaps this is to enable the use of external feedback by outputting the PLL output to a pin and feeding it to the CLKFB pin?

alibabashack, 2018-12-30:
So your point is that the feedback path via two dividers (DIVoutput and DIVfeedback) is unusual, because a separate output divider would allow finer frequency steps at large frequencies. Correct? In this case there are still three unused output dividers left, since only one of the four available dividers is part of the feedback path.