YosysHQ/prjtrellis #49: NextPnr build failure

whatnick, 2018-11-25:
Following build instructions on Ubuntu 16.04 (WSL). Successfully built libtrellis and yosys. Failing on Nextpnr - 

`prjtrellis/libtrellis/src/BitDatabase.cpp:27: Trellis::ConfigBit Trellis::cbit_from_str(const string&): Assertion s[idx] == 'F'' failed.`

daveshah1, 2018-11-25:
Thanks for reporting!

I have set up Ubuntu on WSL but cannot reproduce this here. Please could you share exactly how you cloned and built Trellis and nextpnr; as I suspect this may be CRLF related.

whatnick, 2018-11-25:
Thanks for checking this David. I have cloned pjtrellis and done a make and
make install. The same for yosys. I then pointed nextpnr to the cloned
instance of pjtrellis and attempted to build and got this error. If you
believe this is crlf related I can batch run dos2unix and try again.
However I don't believe the compiler is that fragile.

To double check I can attempt to build in a VM.

Regards,

Tisham.

daveshah1, 2018-11-25:
The problem here is that CRLFs might be breaking the Trellis database parser if they are in the database files (prjtrellis/database). This would likely be a problem if you cloned Trellis and its database on Windows and line endings changed to CRLF automatically, then used it with Trellis built on Unix expecting LF.

whatnick, 2018-12-20:
Making in Vagrant solved my issues. Closing for now.
[image]

closed 2018-12-20 by whatnick