YosysHQ/prjtrellis #55: Missing support for LFE5-12F devices

rjordans, 2019-01-09:
Hi all,
I was looking forward to try out this project on my FPGA board but I noticed that the LFE5-12F chips are not mentioned yet in the database or tools.

What would need to be done to get it added?  If needed I have hardware and the Diamond tools installed so I may be able to help out but I would need to get some instructions on how to get started on this.  I'm a bit short on time though so it may take a while if it requires active effort from my side.

Cheers,
 Roel

daveshah1, 2019-01-09:
Hi Roel,

There isn't really any 12F silicon. To build a bitstream for the 12F parts, use nextpnr as if you were building for the 25F, then add ` --idcode 0x21111043` to the `ecppack` command line when creating the bitstream itself.

David

rjordans, 2019-01-09:
Ah, interesting...  Thanks for the info, I´ll go ahead and try the tools on my hardware then!

rjordans, 2019-01-09:
To avoid confusion.  Would it be an idea to add the 12F parts as alias to the 25F in the tools?

C-Elegans, 2019-01-09:
Since the 25f and 12f chips share a die, does that mean that you can use all 25k luts on a 12f chip like the iCE40 hx8k and hx4k?

daveshah1, 2019-01-09:
Yes, you can, however I'd rather not have this a highly advertised "feature" of the Trellis flow at this point in time.

PR #127 (2020-03-25): Add '12k' device
Actually a 25k chip with a different IDCODE, fixes #55 and other hacks in existence around the internet.
closed 2019-01-09 by daveshah1