YosysHQ/prjtrellis #62: Problem with example 'tinyfpga_rev2'

bit0fun, 2019-01-22:
When attempting to build the tinyfpga_rev2 example, the following error is thrown on the output

```
ecppack -v morse_out.config morse.bit
terminate called after throwing an instance of 'std::runtime_error'
  what():  tile CIB_R94C46:VCIB_DCU0 does not exist in chip LFE5UM5G-85F
```
I assume this is a simple fix, but I'm still learning the FPGA architecture to be able to fix it myself at this time. I double checked, and the tinyfpga_rev1 compilation works without errors. 

My best guess would be that the configuration used may be off, or some of the code is not compatible. Line 9 of both Makefiles in the tinyfpga_rev directories do use different config files which may explain it.

All tools are the most up to date as of 2019-01-22 (just built everything this morning). Yosys built and passed all tests to boot.

uname -a output:
Linux archbox 4.20.3-arch1-1-ARCH #1 SMP PREEMPT Wed Jan 16 22:38:58 UTC 2019 x86_64 GNU/Linuxiver

daveshah1, 2019-01-22:
The problem is that the example didn't set the device type to the UM5G variant correctly,  which didn't matter when this example was written but does now.

d83ddec fixes that and generally brings the example up to date with the latest feature set.

bit0fun, 2019-01-22:
All works as intended, glad this could be resolved quickly

closed 2019-01-22 by bit0fun