YosysHQ/prjtrellis #75: database for ECP5 CABGA381 is missing pin U3

andres-erbsen, 2019-03-29:
When I try to use pin `U3` in a design on the ECP5 EVN board, nextpnr says `ERROR: IO pin 'sck$tr_io' constrained to pin 'U3', which does not exist for package 'CABGA381'.` And indeed, the pin does not seem to be present in the database:

```
$ grep -P '(BGA|"U3")' /usr/share/trellis/database/ECP5/LFE5UM5G-85F/iodb.json
        "CABGA381": {
        "CABGA554": {
            "U3": {
        "CABGA756": {
            "U3": {
        "CSFBGA285": {
```

However, the datasheet of the evaluation board says that this pin is connected to the clock of the onboard SPI flash chip, and the ecp5 pinout document (https://www.latticesemi.com/view_document?document_id=50487) confirms as well:

```
$ head -5 /tmp/ECP5U85Pinout.csv | tail -1
PAD,Pin/Ball Function,Bank,Dual Function,Differential,High Speed,DQS,CABGA756,CABGA554,CABGA381,CSFBGA285
$ grep 'SCK.*U3' /tmp/ECP5U85Pinout.csv   
201,CCLK,8,MCLK/SCK,-,-,-,AM3,AE3,U3,U16
```

Is this an easy fix on the software side, or should I just go ahead and add a jumper wire to route this signal to some other pin?

// sidenote: thanks for building this, I had a very smooth plug-and-play-and-it-actually-just-works experience so far!

daveshah1, 2019-03-31:
This pin requires a special primitive (USRMCLK) to use it, it cannot be used as a general IO pin and thus does not appear in the database. Work to support it is underway.

closed 2019-03-31 by daveshah1

daveshah1, 2019-11-27:
USRMCLK is now supported in nextpnr.