YosysHQ/prjtrellis #76: Error in `openocd': double free or corruption (!prev): 0x000000000246d130

RCoeurjoly, 2019-03-31:
With a fresh install, I try to build the ecp5_env example:

`make prog`

I get the following error:

> *** Error in `openocd': double free or corruption (!prev): 0x0000000001491ff0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f94cf8337e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f94cf83c37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f94cf84053c]
openocd[0x44e741]
openocd[0x450250]
openocd[0x448d05]
/usr/lib/x86_64-linux-gnu/libjim.so.0.76(+0x199b8)[0x7f94cfea89b8]
======= Memory map: ========
00400000-005c7000 r-xp 00000000 fd:00 3739877                            /usr/bin/openocd
007c6000-007c7000 r--p 001c6000 fd:00 3739877                            /usr/bin/openocd
007c7000-007d0000 rw-p 001c7000 fd:00 3739877                            /usr/bin/openocd
007d0000-0084b000 rw-p 00000000 00:00 0 
01411000-0161b000 rw-p 00000000 00:00 0                                  [heap]
7f94c8000000-7f94c8021000 rw-p 00000000 00:00 0 
7f94c8021000-7f94cc000000 ---p 00000000 00:00 0 
7f94ce179000-7f94ce18f000 r-xp 00000000 fd:00 9044033                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f94ce18f000-7f94ce38e000 ---p 00016000 fd:00 9044033                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f94ce38e000-7f94ce38f000 rw-p 00015000 fd:00 9044033                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f94ce38f000-7f94ce992000 rw-p 00000000 00:00 0 
7f94ce992000-7f94ce993000 ---p 00000000 00:00 0 
7f94ce993000-7f94cf193000 rw-p 00000000 00:00 0 
7f94cf193000-7f94cf19a000 r-xp 00000000 fd:00 9055528                    /lib/x86_64-linux-gnu/librt-2.23.so
7f94cf19a000-7f94cf399000 ---p 00007000 fd:00 9055528                    /lib/x86_64-linux-gnu/librt-2.23.so
7f94cf399000-7f94cf39a000 r--p 00006000 fd:00 9055528                    /lib/x86_64-linux-gnu/librt-2.23.so
7f94cf39a000-7f94cf39b000 rw-p 00007000 fd:00 9055528                    /lib/x86_64-linux-gnu/librt-2.23.so
7f94cf39b000-7f94cf39e000 r-xp 00000000 fd:00 9055530                    /lib/x86_64-linux-gnu/libdl-2.23.so
7f94cf39e000-7f94cf59d000 ---p 00003000 fd:00 9055530                    /lib/x86_64-linux-gnu/libdl-2.23.so
7f94cf59d000-7f94cf59e000 r--p 00002000 fd:00 9055530                    /lib/x86_64-linux-gnu/libdl-2.23.so
7f94cf59e000-7f94cf59f000 rw-p 00003000 fd:00 9055530                    /lib/x86_64-linux-gnu/libdl-2.23.so
7f94cf59f000-7f94cf5b7000 r-xp 00000000 fd:00 9055532                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f94cf5b7000-7f94cf7b6000 ---p 00018000 fd:00 9055532                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f94cf7b6000-7f94cf7b7000 r--p 00017000 fd:00 9055532                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f94cf7b7000-7f94cf7b8000 rw-p 00018000 fd:00 9055532                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f94cf7b8000-7f94cf7bc000 rw-p 00000000 00:00 0 
7f94cf7bc000-7f94cf97c000 r-xp 00000000 fd:00 9055545                    /lib/x86_64-linux-gnu/libc-2.23.so
7f94cf97c000-7f94cfb7c000 ---p 001c0000 fd:00 9055545                    /lib/x86_64-linux-gnu/libc-2.23.so
7f94cfb7c000-7f94cfb80000 r--p 001c0000 fd:00 9055545                    /lib/x86_64-linux-gnu/libc-2.23.so
7f94cfb80000-7f94cfb82000 rw-p 001c4000 fd:00 9055545                    /lib/x86_64-linux-gnu/libc-2.23.so
7f94cfb82000-7f94cfb86000 rw-p 00000000 00:00 0 
7f94cfb86000-7f94cfc8e000 r-xp 00000000 fd:00 9055548                    /lib/x86_64-linux-gnu/libm-2.23.so
7f94cfc8e000-7f94cfe8d000 ---p 00108000 fd:00 9055548                    /lib/x86_64-linux-gnu/libm-2.23.so
7f94cfe8d000-7f94cfe8e000 r--p 00107000 fd:00 9055548                    /lib/x86_64-linux-gnu/libm-2.23.so
7f94cfe8e000-7f94cfe8f000 rw-p 00108000 fd:00 9055548                    /lib/x86_64-linux-gnu/libm-2.23.so
7f94cfe8f000-7f94cfece000 r-xp 00000000 fd:00 3739874                    /usr/lib/x86_64-linux-gnu/libjim.so.0.76
7f94cfece000-7f94d00ce000 ---p 0003f000 fd:00 3739874                    /usr/lib/x86_64-linux-gnu/libjim.so.0.76
7f94d00ce000-7f94d00d1000 r--p 0003f000 fd:00 3739874                    /usr/lib/x86_64-linux-gnu/libjim.so.0.76
7f94d00d1000-7f94d00d2000 rw-p 00042000 fd:00 3739874                    /usr/lib/x86_64-linux-gnu/libjim.so.0.76
7f94d00d2000-7f94d00e9000 r-xp 00000000 fd:00 9044263                    /lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0
7f94d00e9000-7f94d02e8000 ---p 00017000 fd:00 9044263                    /lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0
7f94d02e8000-7f94d02e9000 r--p 00016000 fd:00 9044263                    /lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0
7f94d02e9000-7f94d02ea000 rw-p 00017000 fd:00 9044263                    /lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0
7f94d02ea000-7f94d02f1000 r-xp 00000000 fd:00 9044251                    /lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4
7f94d02f1000-7f94d04f0000 ---p 00007000 fd:00 9044251                    /lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4
7f94d04f0000-7f94d04f1000 r--p 00006000 fd:00 9044251                    /lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4
7f94d04f1000-7f94d04f2000 rw-p 00007000 fd:00 9044251                    /lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4
7f94d04f2000-7f94d04f3000 rw-p 00000000 00:00 0 
7f94d04f3000-7f94d04f6000 r-xp 00000000 fd:00 3739872                    /usr/lib/x86_64-linux-gnu/libhidapi-hidraw.so.0.0.0
7f94d04f6000-7f94d06f5000 ---p 00003000 fd:00 3739872                    /usr/lib/x86_64-linux-gnu/libhidapi-hidraw.so.0.0.0
7f94d06f5000-7f94d06f6000 r--p 00002000 fd:00 3739872                    /usr/lib/x86_64-linux-gnu/libhidapi-hidraw.so.0.0.0
7f94d06f6000-7f94d06f7000 rw-p 00003000 fd:00 3739872                    /usr/lib/x86_64-linux-gnu/libhidapi-hidraw.so.0.0.0
7f94d06f7000-7f94d071d000 r-xp 00000000 fd:00 9055531                    /lib/x86_64-linux-gnu/ld-2.23.so
7f94d08cb000-7f94d08cf000 rw-p 00000000 00:00 0 
7f94d08cf000-7f94d08ed000 r-xp 00000000 fd:00 9044079                    /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f94d08ed000-7f94d08ee000 r--p 0001d000 fd:00 9044079                    /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f94d08ee000-7f94d08ef000 rw-p 0001e000 fd:00 9044079                    /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f94d08ef000-7f94d08f2000 rw-p 00000000 00:00 0 
7f94d091b000-7f94d091c000 rw-p 00000000 00:00 0 
7f94d091c000-7f94d091d000 r--p 00025000 fd:00 9055531                    /lib/x86_64-linux-gnu/ld-2.23.so
7f94d091d000-7f94d091e000 rw-p 00026000 fd:00 9055531                    /lib/x86_64-linux-gnu/ld-2.23.so
7f94d091e000-7f94d091f000 rw-p 00000000 00:00 0 
7ffcb5a14000-7ffcb5a35000 rw-p 00000000 00:00 0                          [stack]
7ffcb5b6c000-7ffcb5b6f000 r--p 00000000 00:00 0                          [vvar]
7ffcb5b6f000-7ffcb5b70000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Makefile:18: recipe for target 'prog' failed
make: *** [prog] Aborted

I think I have install all dependencies correctly.

Hardware used is  Lattice ECP5 5G Evaluation Board RevB (P/N: LEF5UM5G-85F-EVN).

The board has the jumper is JP2.

daveshah1, 2019-03-31:
This bug is almost certainly in openocd. Are you using the latest version of it, or an older version (perhaps a distro-provided one?)

RCoeurjoly, 2019-04-01:
True enough, I was using 0.9.0, provided by my distro (Trisquel 8.0).
Then I installed the latest version from source, following the installation instructions provided by openocd.
I got an error saying that ftdi was not a valid interface.

I finally install openocd with:
`./configure --enable-ftdi`

And everything works now.

Maybe in other distros it is enable by default?

Otherwise, we could provide an indication of enabling the flag in the installation instructions.

daveshah1, 2019-04-02:
As far as I know, this is usually enabled by default, but added a note in 7848ab8 for anyone building it themselves.

closed 2019-04-02 by daveshah1