YosysHQ/prjtrellis #83: RISCV example not working

BracketMaster, 2019-07-29:
Programming the RISCV example into the FPGA does not do anything. The LED doesn't blink and the UART doesn't send. I investigated this three months ago and determined that Lattice incorrectly routed the UART signals from the FTDI chip on the PCB. Soldering R34 and R35 on the evaluation board result in faint FTDI signals. I talked to another ECP5 user who had the same problem. For. the evaluation board, the only solution I found was to connect an FTDI cable directly too pins.

Concerning the LED, I can't remember the exact issue (as I fixed it months ago) - but basically there was an issue with the binary being generated.

I fixed the example [here](https://github.com/BracketMaster/ECP5-RISCV32-with-C-and-UART-Example). It has some extra goodies like an iverilog test bench. I was considering correcting the RISCV example here - but I'm wondering if a submodule might be more appropriate.

trabucayre, 2019-11-11:
I've just received an ecp5 evn board. I have the same issue with UART. No problem with leds.
I have dumped the eeprom, used to configure ft2232h device, using libftdi.
INTERFACE_B (BxyBUSz) is configured in FIFO mode (0x01 in byte 0x01) instead of UART (0x00 in byte 0x01). This is why UART is not working.
I'm, currently, not able to use the libftdi to rebuild eeprom content, so I've done an ugly hack to change value, compute new checksum and write manually modified eeprom.
As soon as I have working piece of code, I store this somewhere in internet and post the link.

BracketMaster, 2019-11-11:
Oh. Cool!

trabucayre, 2019-11-12:
I have creates https://github.com/trabucayre/fixFT2232_ecp5evn with a dedicated application to fix the configuration of my board.
It may be good to backup (with ftdi_eeprom) the official configuration for security.

janrinze, 2019-11-23:
If i understand correctly this does not affect the ability to upload binaries to the ecp5evn?
Having the second channel available as a proper serial port is a very good idea i.m.h.o.
Will try it out when i have a backup of the eeprom.

trabucayre, 2019-11-23:
FT2232 has two independant interfaces (A and B). Each of them has dedicated, independant, area in eeprom to configure all parameters.
See https://www.intra2net.com/en/developer/libftdi/documentation/ftdi_8c_source.html around line 3600.
The eeprom backup is theorically not mandatory, but I prefer specify this step for a bit of security.

BracketMaster, 2020-01-11:
So if I understand correctly, doing ``./fixFT2232_ecp5evn -v 0x403 -p 0x601`` will modify ``INTERFACE_B`` to work in UART mode? Is this correct

trabucayre, 2020-01-12:
Hi,
Yes it's exactly the goal of this app.

ipacman, 2020-04-18:
Ok, thought I'd share my experience in resolving this. 

For the Lattice ECP5-5G EVN board, there are THREE problems:

(1) The Ft2232h needs to have port B programmed for Virtual Com port using the FTDI FT_PROG utility. Yeah, I looked for a Linux version of this but ended up running FT_PROG (once) on Windows just to get this configuration changed. 

Its default driver setting is D2XX. In the hardware section, I also changed to "RS232" for port B.

Refer to https://numato.com/kb/configuring-ft2232h-usb-serial-converter-saturn-spartan-6-module

(2) Of course, R34 and R35 needs to be "installed". They are both 0 ohm resistors, so I used a bit of wire. Thats clear from the schematic -- the FTDI pins used for UART are otherwise disconnected.

(3) But It still didnt work (for me) and sounds like for others.

I was nervous when I looked at the schematic because these wires run all over town. It turns out 0 ohm R22 and R23 need to be removed because "something else" on the signals are pulling the lines down so the voltage doesnt get high enough for the FT2232H chip. Thats a bad scene anyway, so removing the resistors frees the signals to work right. Referring to page 7, we see the signals are wired to D7 and E6 of the fpga which *must* be grounded (the example doesnt set them). 

Another solution for #3 is to insure the fpga wasnt grounding unused signals but I havent learned the tools enough yet to know where that is set.

Lattice must have thought no one would be using the UART mode and instead configured it for I3S or something.