YosysHQ/prjtrellis #85: LFE5UM5G-85F is missing pins REFCLKP_D1 and REFCLKN_D1

DurandA, 2019-08-03:
The database for _LFE5UM5G-85F_ is missing the pins `REFCLKP_D1` (Y19 on _CABGA381 package_) and `REFCLKN_D1` (W20). I used the [ECP5UM5G-85 Pinout](http://www.latticesemi.com/view_document?document_id=51579) document as a reference.

In the [ECP5 Evaluation Board](http://www.latticesemi.com/ecp5-evaluation), these pins provide a 200 Mhz clock.

DurandA, 2019-08-03:
Is it related to #75? If so, I guess I will close the issue.

Is there a way to use this pin yet?

daveshah1, 2019-08-05:
These aren't normal IO pins so don't really make sense in the database in a normal way.

You can use them in nextpnr as reference clock inputs to the SERDES like https://github.com/whitequark/Yumewatari/blob/master/yumewatari/gateware/platform/lattice_ecp5.py#L35-L42 (this uses the constraint of the SERDES so doesn't care about the clock pins themselves)

Use of the reference clock without a SERDES isn't supported in nextpnr, but that's really a separate issue that should be filed against nextpnr if Diamond allows this kind of usage.

DurandA, 2019-08-05:
Thanks for the clarification!

closed 2019-08-05 by DurandA