YosysHQ/prjtrellis #93: Versa5g example not working

whit2333, 2019-09-06:
I just got a versa dev kit and tried out the versa5g example. I am getting the following error:

```
openocd -f /usr/local/share/trellis/misc/openocd/ecp5-versa5g.cfg -c "transport select jtag; init; svf demo.svf; exit"
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 25000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Warn : Transport "jtag" was already selected
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 25000 kHz
Info : JTAG tap: ecp5.tap tap/device found: 0x01112043 (mfg: 0x021 (Lattice Semi.), part: 0x1112, ver: 0x0)
Warn : JTAG tap: ecp5.tap       UNEXPECTED: 0x01112043 (mfg: 0x021 (Lattice Semi.), part: 0x1112, ver: 0x0)
Error: JTAG tap: ecp5.tap  expected 1 of 1: 0x81112043 (mfg: 0x021 (Lattice Semi.), part: 0x1112, ver: 0x8)
Error: Trying to use configured scan chain anyway...
Warn : Bypassing JTAG setup events due to errors
Warn : gdb services need one or more targets defined
svf processing file: "demo.svf"
HDR	0;
HIR	0;
TDR	0;
TIR	0;
ENDDR	DRPAUSE;
ENDIR	IRPAUSE;
STATE	IDLE;
SIR	8	TDI  (E0);
			MASK (FFFFFFFF);
SIR	8	TDI  (1C);
				FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF);
SIR	8	TDI  (C6);
SDR	8	TDI  (00);
RUNTEST	IDLE	2 TCK	1.00E-02 SEC;
SIR	8	TDI  (3C);
			MASK (0000B000);
SIR	8	TDI  (46);
SDR	8	TDI  (01);
RUNTEST	IDLE	2 TCK	1.00E-02 SEC;
SIR	8	TDI  (7A);
RUNTEST	IDLE	2 TCK	1.00E-02 SEC;
			FFFFFFFFCDBDFFFFFF008C1CCC82E24282C21CB462AC2CB4E2ACB2AAACA26232045C2E4E860A00FF);
			40040040000000000000000000000000000000000000000000000000000000000000000000000000);
			0000000000000000000000FFA1850000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000010060060000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			000000000000000000000000000000FF304400000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000FF3044000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			0000000000000000000000000000000000000000000000FF30440000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			000000000000000000000000000000000000000000000000000000FF304400000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000FF3044000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			0000000000000000000000000000000000000000000000000000000000000000000000FF30440000);
			FF304400000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			000000000000000000000000000000000000000000000000000000000000000000000000000000FF);
			00000000FF3044000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			0000000000000000FF30440000000000000000000000000000000000000000000000000000000000);
[... 54 lines cut ...]
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
SIR	8	TDI  (FF);
RUNTEST	IDLE	100 TCK	1.00E-02 SEC;
SIR	8	TDI  (C0);
RUNTEST	IDLE	2 TCK	1.00E-03 SEC;
			MASK (FFFFFFFF);
SIR	8	TDI  (26);
RUNTEST	IDLE	2 TCK	2.00E-01 SEC;
SIR	8	TDI  (FF);
RUNTEST	IDLE	2 TCK	1.00E-03 SEC;
SIR	8	TDI  (3C);
			MASK (00002100);
Error: tdo check error at line 11
Error:     READ = 0x1112043
Error:     WANT = 0x81112043
Error:     MASK = 0xffffffff

Time used: 0m1s38ms 
svf file programmed failed

Makefile:22: recipe for target 'prog' failed
make: *** [prog] Error 1
```

What am I missing?

daveshah1, 2019-09-06:
Sounds like your "Versa 5G" actually has a non-5G ECP5 (there was a bad batch of these a while ago)

whit2333, 2019-09-06:
Oh, I think you are right. Are there any examples for this board that should work?

whit2333, 2019-09-09:
I think I got things working by replacing `5g` with nothing in the `Makefile`.

closed 2019-09-09 by whit2333