YosysHQ/prjtrellis #96: Trouble with CMake / Python Boost

lolsborn, 2019-09-13:
OS: Ubuntu 19.04
Boost: 1.71.0
CMake: 3.13.4
libtrellis: Master

Initially I tried installing libboost via apt, but nothing I tried seemed to install boost python3 libs that satisfied Cmake.  So I decided install boost from source and then link trellis against it.  Cmake seems to be happier initially, but the generated make-file seems to indicate that the python-boost module was not found even though cmake seems to find it.

Would appreciate any insight or  tips for installing trellis on Ubuntu, so far I've just had a lot of heartbreak.

## Cmake Output
```
steven@goron:~/Projects/prjtrellis/libtrellis$ cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DBOOST_ROOT=/usr/local -v .
-- Boost  found.
-- Found Boost components:
   filesystem;thread
-- Boost  found.
-- Found Boost components:
   python-py373;filesystem;thread
-- Boost  found.
-- Found Boost components:
   python-py37;filesystem;thread
-- Boost  found.
-- Found Boost components:
   python-py3;filesystem;thread
-- Boost  found.
-- Found Boost components:
   python3;filesystem;thread
-- Boost  found.
-- Found Boost components:
   python36;filesystem;thread
-- Boost  found.
-- Found Boost components:
   python37;filesystem;thread
-- Boost  found.
-- Found Boost components:
   program_options
-- Configuring done
-- Generating done
-- Build files have been written to: /home/steven/Projects/prjtrellis/libtrellis
```
## Make Error
```
steven@goron:~/Projects/prjtrellis/libtrellis$ make
CMakeFiles/trellis.dir/build.make:262: *** target pattern contains no '%'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:220: CMakeFiles/trellis.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
## Snippet from build.make
```
# External object files for target trellis
trellis_EXTERNAL_OBJECTS =

libtrellis.so: CMakeFiles/trellis.dir/src/Bels.cpp.o
libtrellis.so: CMakeFiles/trellis.dir/src/BitDatabase.cpp.o
libtrellis.so: CMakeFiles/trellis.dir/src/Bitstream.cpp.o
libtrellis.so: CMakeFiles/trellis.dir/src/CRAM.cpp.o
libtrellis.so: CMakeFiles/trellis.dir/src/Chip.cpp.o
libtrellis.so: CMakeFiles/trellis.dir/src/ChipConfig.cpp.o
libtrellis.so: CMakeFiles/trellis.dir/src/Database.cpp.o
libtrellis.so: CMakeFiles/trellis.dir/src/DedupChipdb.cpp.o
libtrellis.so: CMakeFiles/trellis.dir/src/PyTrellis.cpp.o
libtrellis.so: CMakeFiles/trellis.dir/src/RoutingGraph.cpp.o
libtrellis.so: CMakeFiles/trellis.dir/src/Tile.cpp.o
libtrellis.so: CMakeFiles/trellis.dir/src/TileConfig.cpp.o
libtrellis.so: CMakeFiles/trellis.dir/src/Util.cpp.o
libtrellis.so: CMakeFiles/trellis.dir/build.make
libtrellis.so: Boost::python-NOTFOUND
libtrellis.so: /usr/local/lib/libboost_filesystem.so.1.71.0
libtrellis.so: /usr/local/lib/libboost_thread.so.1.71.0
libtrellis.so: /usr/lib/x86_64-linux-gnu/libpython3.7m.so
libtrellis.so: CMakeFiles/trellis.dir/link.txt
        @$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --green --bold --progress-dir=/home/steven/Projects/prjtrellis/libtrellis/CMakeFiles --progress-num=$(CMAKE_PROGRESS_14) "Linking CXX shared library libtrellis.so"
        $(CMAKE_COMMAND) -E cmake_link_script CMakeFiles/trellis.dir/link.txt --verbose=$(VERBOSE)
```

daveshah1, 2019-09-13:
Just to make sure it's not lingering files from when Boost Python wasn't installef, can you try `rm -rf Makefile CMakeCache.txt CMakeFiles` and then cmake and make again?

daveshah1, 2019-09-13:
If possible now, I would also suggest removing manually installed Boost, and trying again with the distro package (because CMake has to try different varieties of Boost Python, some "not found" warnings are fine so long as it finds it eventually. Finding all the different varieties definitely looks wrong).

lolsborn, 2019-09-13:
I'm still not sure what is going on with my system boost, I tried re-installing it via apt and the header files end up missing.  I think it has something to do with apt meta-packages, but I gave up.

I ended up commenting out the entire section checking for python libs and just added python to the boost_libs variable like below and it seems to have built fine.

```
-set(boost_libs filesystem thread)
+set(boost_libs filesystem python thread)
 
 find_package(PythonInterp 3.5 REQUIRED)
 
@@ -33,52 +33,6 @@ endif()
 
 find_package(Boost REQUIRED COMPONENTS ${boost_libs})
 
-if (BUILD_PYTHON)
-    # Find Boost::Python of a suitable version in a cross-platform way
-    # Some distributions (Arch) call it libboost_python3, others such as Ubuntu
-    # call it libboost_python35. In the latter case we must consider all minor versions
-    # Original source: https://github.com/BVLC/caffe/blob/master/cmake/Dependencies.cmake#L148
-    set(version ${PYTHONLIBS_VERSION_STRING})
-
-    STRING(REGEX REPLACE "[^0-9]" "" boost_py_version "${version}")
-    find_package(Boost COMPONENTS "python-py${boost_py_version}" ${boost_libs})
-    set(Boost_PYTHON_FOUND ${Boost_PYTHON-PY${boost_py_version}_FOUND})
-
-    while (NOT "${version}" STREQUAL "" AND NOT Boost_PYTHON_FOUND)
-        STRING(REGEX REPLACE "([0-9.]+).[0-9]+" "\\1" version "${version}")
-
-        STRING(REGEX REPLACE "[^0-9]" "" boost_py_version "${version}")
```

lolsborn, 2019-09-14:
I'm still not really certain what the problem is/was on that machine.  Now that I'm home I'm building on my PC running 18.04 and system boost and it's building with no troubles. I'm curious if it's something with 19.04 or just (more likely) something weird going on with my system.

I'll close this as I have things working now, but if I can determine what was really the core of my issue before let you know some time down the road.

closed 2019-09-14 by lolsborn

tommythorn, 2020-03-20:
I'm on Ubuntu 19.10 and there's no `boost-python`.  I haven't any packages to get me past
```
CMake Error at CMakeLists.txt:84 (message):
  No version of Boost::Python 3.x could be found.
```

daveshah1, 2020-03-20:
Try either `libboost-all-dev` or just `libboost-python-dev` if you prefer

Chandler-Kluser, 2023-06-10:
> Try either `libboost-all-dev` or just `libboost-python-dev` if you prefer

Thanks a lot, `libboost-all-dev` solved the issue!!