YosysHQ/prjtrellis #98: `TypeError: __init__() should return None, not 'NoneType'` on Mac

xobs, 2019-09-16:
When compiling on a Mac using Conda, the resulting `pytrellis.so` cannot be used.

The library has correct bindings to `libboost_python3.7.dylib` and to `libpython3.7m.dylib`, however it gives the error `TypeError: __init__() should return None, not 'NoneType'` as soon as I try to instantiate a `Chip`:

```sh
Builders-iMac:libtrellis builder$ otool -L pytrellis.so
pytrellis.so:
        @rpath/libboost_python37.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libboost_filesystem.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libboost_thread.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libboost_system.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libboost_chrono.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libboost_date_time.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libboost_atomic.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libpython3.7m.dylib (compatibility version 3.7.0, current version 3.7.0)
        /usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 400.9.4)
        /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.250.1)
Builders-iMac:libtrellis builder$ PYTHONPATH=/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/libtrellis:/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/util/common:/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/timing/util /tmp/nextpnr/bin/python3.7 -c "import pytrellis; import database; pytrellis.load_database(database.get_db_root()); pytrellis.Chip('LFE5UM5G-25F')"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
TypeError: __init__() should return None, not 'NoneType'
Builders-iMac:libtrellis builder$
```

This is using the correct version of Boost::Python, coupled with the correct version of Python.  Changing to use static Boost libraries does not affect this, however building the library statically does cause a segfault.

daveshah1, 2019-09-16:
I haven't seen this error on Linux before. Just to check, did you see https://stackoverflow.com/questions/20437769/boost-python-init-should-return-none-not-nonetype?

xobs, 2019-09-16:
That was my first thought, which is why I've posted the output of `otool -L`, confirming that `pytrellis.so` is linked against the correct library.

If I remove a random object file to force a rebuild and a relink, you can see that it's linking against the correct library:

```
[ 52%] Building CXX object CMakeFiles/pytrellis.dir/src/TileConfig.cpp.o
/Library/Developer/CommandLineTools/usr/bin/c++  -DINCLUDE_PYTHON=1 -Dpytrellis_EXPORTS -I/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/libtrellis/include -I/tmp/nextpnr/include -I/tmp/nextpnr/include/python3.7m  -Wall -pedantic -Wextra -O3 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -fPIC   -std=gnu++14 -o CMakeFiles/pytrellis.dir/src/TileConfig.cpp.o -c /Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/libtrellis/src/TileConfig.cpp
[ 55%] Linking CXX shared module pytrellis.so
/Users/builder/Documents/homebrew/Cellar/cmake/3.15.3/bin/cmake -E cmake_link_script CMakeFiles/pytrellis.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/c++ -Wall -pedantic -Wextra -O3 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -bundle -Wl,-headerpad_max_install_names  -o pytrellis.so CMakeFiles/pytrellis.dir/src/Bels.cpp.o CMakeFiles/pytrellis.dir/src/BitDatabase.cpp.o CMakeFiles/pytrellis.dir/src/Bitstream.cpp.o CMakeFiles/pytrellis.dir/src/CRAM.cpp.o CMakeFiles/pytrellis.dir/src/Chip.cpp.o CMakeFiles/pytrellis.dir/src/ChipConfig.cpp.o CMakeFiles/pytrellis.dir/src/Database.cpp.o CMakeFiles/pytrellis.dir/src/DedupChipdb.cpp.o CMakeFiles/pytrellis.dir/src/PyTrellis.cpp.o CMakeFiles/pytrellis.dir/src/RoutingGraph.cpp.o CMakeFiles/pytrellis.dir/src/Tile.cpp.o CMakeFiles/pytrellis.dir/src/TileConfig.cpp.o CMakeFiles/pytrellis.dir/src/Util.cpp.o -Wl,-rpath,/tmp/nextpnr/lib /tmp/nextpnr/lib/libboost_python37.dylib /tmp/nextpnr/lib/libboost_filesystem.dylib /tmp/nextpnr/lib/libboost_thread.dylib /tmp/nextpnr/lib/libboost_system.dylib /tmp/nextpnr/lib/libboost_chrono.dylib /tmp/nextpnr/lib/libboost_date_time.dylib /tmp/nextpnr/lib/libboost_atomic.dylib /tmp/nextpnr/lib/libpython3.7m.dylib
```

daveshah1, 2019-09-16:
Do you think you could try a patch like https://github.com/microsoft/bond/pull/249/files (but for 3.7)? This is reported to fix a similar issue elsewhere.

xobs, 2019-09-16:
I'm specifying the python library and the path to python in my `cmake` command:

```
    cmake \
        -DBUILD_SHARED=ON \
        -DSTATIC_BUILD=OFF \
        -DBUILD_PYTHON=ON \
        -DBoost_USE_STATIC_LIBS=OFF \
        -DBOOST_ROOT=/tmp/nextpnr \
        -DCMAKE_EXE_LINKER_FLAGS='-fno-lto -ldl -lutil' \
        -DPYTHON_LIBRARY=/tmp/nextpnr/lib/libpython3.7m.dylib \
        -DPYTHON_EXECUTABLE=/tmp/nextpnr/bin/python3.7 \
        .
```

That patch appears to be for auto-discovering the python library, and so wouldn't get triggered in my configuration.

daveshah1, 2019-09-16:
Is there any way that Boost could have been built for a different Python library?

xobs, 2019-09-16:
It's possible, but these files are all from the `anaconda` repo, and all appear to be for python 3.7:

```sh
$ ls -l build-data/darwin/
-rw-r--r--  1 builder  staff     12001 Sep 16 17:44 boost-1.67.0-py37_4.tar.bz2
-rw-r--r--  1 builder  staff    992738 Sep 16 17:44 eigen3-3.3.7-0.tar.bz2
-rw-r--r--  1 builder  staff  19723983 Sep 16 17:44 libboost-1.67.0-hebc422b_4.tar.bz2
-rw-r--r--  1 builder  staff    324669 Sep 16 17:44 py-boost-1.67.0-py37h6440ff4_4.tar.bz2
-rw-r--r--  1 builder  staff  22698124 Sep 16 17:44 python-3.7.4-h359304d_1.tar.bz2
$
```

daveshah1, 2019-09-16:
And have you double checked that the interpreter actually running is the Conda one and not the system one?

xobs, 2019-09-16:
Yes, you can see when I reproduce the case above I'm running /tmp/nextpnr/bin/python3.7 directly.

xobs, 2019-09-16:
I switched to python3.6, to see if maybe that was causing issues, but I got the same result.

In fact, I can see that it appears to be loading all the correct libraries, it's just not working:

```sh
Builders-iMac:nextpnr builder$ DYLD_PRINT_LIBRARIES=1 PYTHONPATH=/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/libtrellis:/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/util/common:/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/timing/util /tmp/nextpnr/bin/python3 /Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/nextpnr/ecp5/trellis_import.py -p /Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/nextpnr/ecp5/constids.inc 25k
dyld: loaded: /tmp/nextpnr/bin/python3
dyld: loaded: /usr/lib/libSystem.B.dylib
dyld: loaded: /usr/lib/system/libcache.dylib
dyld: loaded: /usr/lib/system/libcommonCrypto.dylib
dyld: loaded: /usr/lib/system/libcompiler_rt.dylib
dyld: loaded: /usr/lib/system/libcopyfile.dylib
dyld: loaded: /usr/lib/system/libcorecrypto.dylib
dyld: loaded: /usr/lib/system/libdispatch.dylib
dyld: loaded: /usr/lib/system/libdyld.dylib
dyld: loaded: /usr/lib/system/libkeymgr.dylib
dyld: loaded: /usr/lib/system/liblaunch.dylib
dyld: loaded: /usr/lib/system/libmacho.dylib
dyld: loaded: /usr/lib/system/libquarantine.dylib
dyld: loaded: /usr/lib/system/libremovefile.dylib
dyld: loaded: /usr/lib/system/libsystem_asl.dylib
dyld: loaded: /usr/lib/system/libsystem_blocks.dylib
dyld: loaded: /usr/lib/system/libsystem_c.dylib
dyld: loaded: /usr/lib/system/libsystem_configuration.dylib
dyld: loaded: /usr/lib/system/libsystem_coreservices.dylib
dyld: loaded: /usr/lib/system/libsystem_darwin.dylib
dyld: loaded: /usr/lib/system/libsystem_dnssd.dylib
dyld: loaded: /usr/lib/system/libsystem_info.dylib
dyld: loaded: /usr/lib/system/libsystem_m.dylib
dyld: loaded: /usr/lib/system/libsystem_malloc.dylib
dyld: loaded: /usr/lib/system/libsystem_networkextension.dylib
dyld: loaded: /usr/lib/system/libsystem_notify.dylib
dyld: loaded: /usr/lib/system/libsystem_sandbox.dylib
dyld: loaded: /usr/lib/system/libsystem_secinit.dylib
dyld: loaded: /usr/lib/system/libsystem_kernel.dylib
dyld: loaded: /usr/lib/system/libsystem_platform.dylib
dyld: loaded: /usr/lib/system/libsystem_pthread.dylib
dyld: loaded: /usr/lib/system/libsystem_symptoms.dylib
dyld: loaded: /usr/lib/system/libsystem_trace.dylib
dyld: loaded: /usr/lib/system/libunwind.dylib
dyld: loaded: /usr/lib/system/libxpc.dylib
dyld: loaded: /usr/lib/libobjc.A.dylib
dyld: loaded: /usr/lib/libc++abi.dylib
dyld: loaded: /usr/lib/libc++.1.dylib
dyld: loaded: /tmp/nextpnr/lib/python3.6/lib-dynload/_heapq.cpython-36m-darwin.so
dyld: loaded: /Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/libtrellis/pytrellis.so
dyld: loaded: /tmp/nextpnr/lib/libboost_python36.dylib
dyld: loaded: /tmp/nextpnr/lib/libboost_filesystem.dylib
dyld: loaded: /tmp/nextpnr/lib/libboost_thread.dylib
dyld: loaded: /tmp/nextpnr/lib/libboost_system.dylib
dyld: loaded: /tmp/nextpnr/lib/libboost_chrono.dylib
dyld: loaded: /tmp/nextpnr/lib/libboost_date_time.dylib
dyld: loaded: /tmp/nextpnr/lib/libboost_atomic.dylib
dyld: loaded: /tmp/nextpnr/lib/libpython3.6m.dylib
dyld: loaded: /tmp/nextpnr/lib/python3.6/lib-dynload/_json.cpython-36m-darwin.so
dyld: loaded: /tmp/nextpnr/lib/python3.6/lib-dynload/_posixsubprocess.cpython-36m-darwin.so
dyld: loaded: /tmp/nextpnr/lib/python3.6/lib-dynload/select.cpython-36m-darwin.so
dyld: loaded: /tmp/nextpnr/lib/python3.6/lib-dynload/math.cpython-36m-darwin.so
dyld: loaded: /tmp/nextpnr/lib/python3.6/lib-dynload/_struct.cpython-36m-darwin.so
Traceback (most recent call last):
  File "/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/nextpnr/ecp5/trellis_import.py", line 496, in <module>
    main()
  File "/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/nextpnr/ecp5/trellis_import.py", line 482, in main
    chip = pytrellis.Chip(dev_names[args.device])
TypeError: __init__() should return None, not 'NoneType'
Builders-iMac:nextpnr builder$
```

daveshah1, 2019-09-16:
Shouldn't make a difference, but might be worth moving `pytrellis.so` into `/tmp/nextpnr/lib`?

xobs, 2019-09-16:
Good thinking, but it doesn't make a difference.

It should be noted that it's not instantaneous -- the failure actually takes about half a second:

```sh
Builders-iMac:libtrellis builder$ time PYTHONPATH=/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/libtrellis:/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/util/common:/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/timing/util /tmp/nextpnr/bin/python3.6 -c "import pytrellis; import database; pytrellis.load_database(database.get_db_root()); pytrellis.Chip('LFE5UM5G-25F')"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
TypeError: __init__() should return None, not 'NoneType'

real    0m0.593s
user    0m0.519s
sys     0m0.069s
Builders-iMac:libtrellis builder$ time PYTHONPATH=/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/libtrellis:/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/util/common:/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/timing/util /tmp/nextpnr/bin/python3.6 -c "import pytrellis; import database; "
Builders-iMac:libtrellis builder$ time PYTHONPATH=/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/libtrellis:/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/util/common:/Users/builder/Documents/toolchain-nextpnr-ecp5/_builds/build_darwin/prjtrellis/timing/util /tmp/nextpnr/bin/python3.6 -c "import pytrellis; import database; pytrellis.load_database(database.get_db_root());"

real    0m0.105s
user    0m0.074s
sys     0m0.027s
Builders-iMac:libtrellis builder$
```

So it's doing _something_, and the failure might be within one of the submodules, if there are any.

xobs, 2019-09-16:
I also tried switching from the `anaconda`-provided packages to those provided by `conda-forge`, and got the exact same error: `TypeError: __init__() should return None, not 'NoneType'`

xobs, 2019-09-17:
More information:

I know I'm compiling against the correct boost, since I added an error to Boost depending on which version it was building for, and it hit the one for >=3.

I'm unsure of how to debug this, so I added "print" messages to the Chip initializer, and those do get printed.  So I know it's at least running that, but then some magic happens.

This is the same Python that I embed in nextpnr, so I'm reasonably sure that the Python install is good.

xobs, 2019-09-17:
For my purposes, I'm giving up on this -- I've spent five days trying to get this working, and at this point I'm going to declare it broken on Darwin and work around the problem.

Because `pytrellis` is used in `nextpnr` to generate `.bba` files, and I'm having similar issues with it on Windows, I'm going to add support for building `nextpnr` with pre-generated `.bba` files.  This should work around the long build times on Windows, as well as the inability to build on Darwin, both of which are caused by Python.

fedorkanin, 2024-09-30:
In my opinion, the issue might be that you have two different Python libs in your otool output:
```
@rpath/libboost_python37.dylib (compatibility version 0.0.0, current version 0.0.0)
...
@rpath/libpython3.7m.dylib (compatibility version 3.7.0, current version 3.7.0)
```

I have resolved a similar issue by using `-undefined dynamic lookup` flag and linking only to `-lboost_pythonXXX`, instead of linking to both `-lboost_pythonXXX` and  `-lpythonX.XX`. This took me several days of debugging. Here is the Stack Overflow answer with a similar approach: https://stackoverflow.com/a/61592909/12870671

anthony-walker, 2024-12-24:
I ran into the same issue and `-undefined dynamic\_lookup` as part of my `LDFLAGS` as suggested by @fedorkanin seems to have done the trick in my case.