YosysHQ/prjtrellis #105: Compile fails: Boost-python on Arch

hexagon5un, 2019-10-08:
On Arch Linux, with Boost 1.71.0 installed, I get the following after running cmake:
-- Found PythonLibs: /usr/lib/libpython3.7m.so (found suitable version "3.7.4", minimum required is "3.5") 
-- Found Boost: /usr/lib64/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components:  filesystem thread 
-- Could NOT find Boost: missing: python-py374 (found /usr/lib64/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0"))
... etc. for other python versions.

Indeed, all symptoms look the same as [this issue on Ubuntu](https://github.com/SymbiFlow/prjtrellis/issues/96).

mmicko, 2019-10-08:
Can you please try adding next line to cmake  ?

set(Boost_NO_BOOST_CMAKE ON)

Seams that Boost 1.71 is packing cmake files that are actually not valid. This helped solving issue compiling with same boost version on macOS.

hexagon5un, 2019-10-08:
I got it to compile by changing
`if (BUILD_PYTHON)` to `if (FALSE)` in CMakeLists.txt, bypassing the check the brute-force way.

I'm a cmake noob, where would I put that set() command?

mmicko, 2019-10-08:
Please try placing it in this line : https://github.com/SymbiFlow/prjtrellis/blob/master/libtrellis/CMakeLists.txt#L7

hexagon5un, 2019-10-08:
Responds with warnings, but compiles as if the checks weren't run.

Looking through the code, the whole python-lib detection section seems to be a bit fragile and distro-dependent.  I.e. it all depends on the names of libs rather than their contents or the ability to compile with them.

mmicko, 2019-10-08:
There is not other way right now, and it is very system and boost version dependent that is why it needs to check multiple configurations.
It is ok to give back warnings, important is if it does compiles.

hexagon5un, 2019-10-08:
True.  

My comment was just to point out that a 'safety check' that fails to detect the safe condition, while preventing compilation, may cause confusion with others as it did with me.

Anyway, I'm good with either solution.  Thanks.

closed 2019-10-08 by hexagon5un

daveshah1, 2019-10-08:
Fix committed with some other improvements as 1c49486. In the long term I may look to move to pybind11 or another alternative that will hopefully be less of a PITA than Boost::Python and CMake