YosysHQ/prjtrellis #108: Versa example issue

lolsborn, 2019-10-11:
This seems related to #93 but I'm not certain.

- The messages printed are errrors instead of warnings and the "Unexpected" IDs printed are not the same as in #93 
- I've connected the board to Diamond and the part number that shows up is 0x81112043 which makes me think it is indeed the correct part.
- The proposed solution on that page doesn't seem to solve my problem.

One interesting thing of note is that even though OpenOCD errors out, the board prints "Hello from project trellis on the versa" so it seems like the bitstream was loaded.

I'm using Ubuntu 19.04 with the latest master branch of trellis.

```
openocd -f /usr/share/trellis/misc/openocd/ecp5-versa5g.cfg -c "transport select jtag; init; svf demo.svf; exit"
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 25000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Warn : Transport "jtag" was already selected
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 25000 kHz
Info : JTAG tap: ecp5.tap tap/device found: 0x00191043 (mfg: 0x021 (Lattice Semi.), part: 0x0191, ver: 0x0)
Warn : JTAG tap: ecp5.tap       UNEXPECTED: 0x00191043 (mfg: 0x021 (Lattice Semi.), part: 0x0191, ver: 0x0)
Error: JTAG tap: ecp5.tap  expected 1 of 1: 0x81112043 (mfg: 0x021 (Lattice Semi.), part: 0x1112, ver: 0x8)
Info : JTAG tap: auto0.tap tap/device found: 0x81112043 (mfg: 0x021 (Lattice Semi.), part: 0x1112, ver: 0x8)
Error: Trying to use configured scan chain anyway...
Warn : AUTO auto0.tap - use "jtag newtap auto0 tap -irlen 2 -expected-id 0x81112043"
Error: IR capture error at bit 10, saw 0x3FFFFFFFFFFFFF051D not 0x...3
Warn : Bypassing JTAG setup events due to errors
Warn : gdb services need one or more targets defined
svf processing file: "demo.svf"
HDR	0;
HIR	0;
TDR	0;
TIR	0;
ENDDR	DRPAUSE;
ENDIR	IRPAUSE;
STATE	IDLE;
SIR	8	TDI  (E0);
			MASK (FFFFFFFF);
SIR	8	TDI  (1C);
				FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF);
SIR	8	TDI  (C6);
SDR	8	TDI  (00);
RUNTEST	IDLE	2 TCK	1.00E-02 SEC;
SIR	8	TDI  (3C);
			MASK (0000B000);
SIR	8	TDI  (46);
SDR	8	TDI  (01);
RUNTEST	IDLE	2 TCK	1.00E-02 SEC;
SIR	8	TDI  (7A);
RUNTEST	IDLE	2 TCK	1.00E-02 SEC;
			007F248941000000620000000200000044C204888100000047000000DCFFFFFFFFCDBDFFFFFF00FF);
			00000000000000000000000000000000000000000004004004004004004000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			0000000000FF7FA30000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000008060000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			000000000000000000FF304400000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000FF3044000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			0000000000000000000000000000000000FF90458000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			000000000000000000000000000000000000000000FF679F00000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
[... 60 lines cut ...]
			00000000000000000000000000000000000000000000000000000000000000FF3044000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
			0000000000000000000000000000000000000000000000000000000000000000000000FF30440000);
			FF304400000000000000000000000000000000000000000000000000000000000000000000000000);
			00000000000000000000000000000000000000000000000000000000000000000000000000000000);
SIR	8	TDI  (FF);
RUNTEST	IDLE	100 TCK	1.00E-02 SEC;
SIR	8	TDI  (C0);
RUNTEST	IDLE	2 TCK	1.00E-03 SEC;
			MASK (FFFFFFFF);
SIR	8	TDI  (26);
RUNTEST	IDLE	2 TCK	2.00E-01 SEC;
SIR	8	TDI  (FF);
RUNTEST	IDLE	2 TCK	1.00E-03 SEC;
SIR	8	TDI  (3C);
			MASK (00002100);
Error: tdo check error at line 11
Error:     READ = 0x2224086
Error:     WANT = 0x81112043
Error:     MASK = 0xffffffff

Time used: 0m1s25ms 
svf file programmed failed
```

daveshah1, 2019-10-11:
Did you follow the instruction in the example README to bypass the ispCLOCK? openocd doesn't like it

> If your Versa board is new, you will need to change J50 to bypass the iSPclock. Re-arrange the jumpers to connect pins 1-2 and 3-5 (leaving one jumper spare). See p19 of the Versa Board user guide.

lolsborn, 2019-10-11:
Nope.  That was the issue.  I didn't even look at the README for the individual examples.  Thanks Dave.

closed 2019-10-11 by lolsborn