YosysHQ/prjtrellis #119: PIO bel missing for pin R16 on CSFBGA285 package of LFE5U-25F

zeldin, 2020-02-22:
The pinout CSV from Lattice contains the following for pin R16 and R17:
```
121,PB18A,8,WRITEN,-,-,-,T3,R16,M9
115,PB13A,8,SN/CSN,True_OF_PB13B,-,-,T2,R17,R8
```
The corresponding entries in `iodb.json` are:
```
            "R16": {
                "col": 18,
                "pio": "A",
                "row": 50
            },
            "R17": {
                "col": 13,
                "pio": "A",
                "row": 50
            },
```
which seems consistent (PB18A = pio A in bottom of col 18, PB13A = pio A in bottom of col 13).
However, only the bottom of col 13 seems to contain a `PIOA` bel in the database, the bottom of col 18 is empty!
```
Python 3.6.9 (default, Jan  2 2020, 19:32:20) 
[GCC 9.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pytrellis
>>> import database
>>> pytrellis.load_database(database.get_db_root())
>>> ddrg = pytrellis.make_dedup_chipdb(pytrellis.Chip("LFE5UM5G-25F"))
>>> [ddrg.to_str(bel.name) for bel in ddrg.locationTypes[ddrg.typeAtLocation[pytrellis.Location(13, 50)]].bels]
['PIOA', 'SIOLOGICA', 'PIOB', 'SIOLOGICB']
>>> [ddrg.to_str(bel.name) for bel in ddrg.locationTypes[ddrg.typeAtLocation[pytrellis.Location(18, 50)]].bels]
[]
>>>
```
This makes nextpnr unable to use the location R16.

smunaut, 2020-02-22:
Are you sure your database, prjtreillis and nextpnr are up to date ?

See https://github.com/YosysHQ/nextpnr/issues/290

zeldin, 2020-02-23:
I was using the latest release of prjtrellis (1.0).  It does indeed seem that the issue is fixed on git HEAD.  Thanks!

closed 2020-02-23 by zeldin