YosysHQ/prjtrellis #135: `PREFIX` is not being respected for `LIBDIR`

FFY00, 2020-04-15:
Building with `-DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_INSTALL_LIBDIR=/lib` results in the libraries being installed to `/lib` instead of `/usr/lib`. This is obviously wrong and can result in broken installation, depending on the environment.

FFY00, 2020-04-15:
https://github.com/SymbiFlow/prjtrellis/blob/fc145eec43ec24a9a98571d1621c81f13eace6a4/libtrellis/CMakeLists.txt#L114

smunaut, 2020-05-19:
Did you try without the `/` ?  AFAIK Cmake only adds the install prefix for relative path. Absolute path are take 'as-is'.  (so that you can have a sys conf dir of `/etc` even if your prefix is `/usr` for instance)

FFY00, 2020-05-19:
No, `CMAKE_INSTALL_FULL_LIBDIR` would be used for that. See https://cmake.org/cmake/help/latest/module/GNUInstallDirs.html. See also https://cmake.org/cmake/help/latest/variable/CMAKE_INSTALL_PREFIX.html#variable:CMAKE_INSTALL_PREFIX.

smunaut, 2020-05-19:
That page states :

```
CMAKE_INSTALL_FULL_<dir>: The absolute path generated from the corresponding CMAKE_INSTALL_<dir> value. If the value is not already an absolute path, an absolute path is constructed typically by prepending the value of the CMAKE_INSTALL_PREFIX variable.":
```

Note the "If the value is not already an absolute path".
So I read that as if CMAKE_INSTALL_LIBDIR is already an absolute path, then it will be taken as-is.

FFY00, 2020-05-19:
You're right, sorry. I was too quick reading through, my bad.

closed 2020-05-19 by FFY00