YosysHQ/prjtrellis #138: Missing pins on LFE5UM5G-45F-8BG381C

xobs, 2020-04-30:
There are several pins that are missing from some devices such as `CABGA381`, namely Y11, Y12, W4, W5, Y5, and Y6, all of which are used for PCIe in e.g. https://github.com/enjoy-digital/litex/blob/master/litex/boards/platforms/versa_ecp5.py#L117-L125

smunaut, 2020-04-30:
Those are all dedicated pins to serdes and not usable as general purpose IO.

Even in diamond, you don't setup any `SITE` constrainst for those, instead you have to lock the serdes block to a specific site in clarity to select the one you want to use.

shuffle2, 2020-04-30:
I'm a bit confused why this example doesn't run into the same problem:
https://github.com/enjoy-digital/liteiclink/blob/master/examples/transceiver/versa_ecp5.py
It outputs an lpf with:
```
LOCATE COMP "refclk1_p" SITE "Y19";
LOCATE COMP "refclk1_n" SITE "W20";
LOCATE COMP "refclk_en" SITE "C12";
IOBUF PORT "refclk_en" IO_TYPE=LVCMOS33;
LOCATE COMP "refclk_rst_n" SITE "R1";
IOBUF PORT "refclk_rst_n" IO_TYPE=LVCMOS33;
LOCATE COMP "pcie_tx_p" SITE "W4";
LOCATE COMP "pcie_tx_n" SITE "W5";
LOCATE COMP "pcie_rx_p" SITE "Y5";
LOCATE COMP "pcie_rx_n" SITE "Y6";
```
Is it only Y11/Y12 causing the problem here?

In my code, I am using `'pcie_x1'` as defined by litex for versa_ecp5, which includes Y11. Diamond (via migen) accepts it, however trellis halts with an error when trying to constrain to Y11. It's true that in diamond, the user is not required to explicitly constrain _any_ of the pcie_x1 pins. However, assigning pins to the expected IP block (e.g. EXTREF0, DCU0, etc) doesn't cause it to error out.

If I just remove my explicit usage of pcie pins, I get: `ERROR: DCU must be constrained to a Bel!`
I'm using verilog to instantiate the DCUA, with `/* synthesis LOC=DCU0 CHAN=CH0 */`. I presume I need something else to tell prjtrellis where to locate the DCU? Note, this `DCU0_inst` is just something spit out from lattice's asbgen, and seems somewhat incompatible with the `DCUA` module defined by prjtrellis.

daveshah1, 2020-04-30:
The problem is that the "synthesis" comment is a horrible relic from the dark ages, that Yosys doesn't support. Use a proper Verilog 2001 attribute like `(* LOC="DCU0" *) and it should be fine in both Yosys and Diamond.

shuffle2, 2020-05-04:
Back to original issue:
I _think_ it was because I did both `platform.request` `refclk` and `pcie_x1`, which resulted in Y11/Y12 being included twice in the lpf (even though I only used clk pins from refclk). So, this was probably just user error and can be closed.