YosysHQ/prjtrellis #139: cmake : found suitable version "3.6.8", minimum required is "3.5"

Martoni, 2020-05-19:
Hello,

Cmake tell me I can't compile the library because my cmake version is too new. Is it known bug ?

```
 cmake -DCMAKE_INSTALL_PREFIX=/usr .
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3.5") 
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
  (Required is at least version "3.5")
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.10/Modules/FindPythonLibs.cmake:262 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:37 (find_package)

-- Configuring incomplete, errors occurred!
See also "/opt/prjtrellis/libtrellis/CMakeFiles/CMakeOutput.log".
```

My Cmake version : 
```
$ cmake --version
cmake version 3.10.2
```
And python3 version : 
```
$ python3 --version
Python 3.6.8
```

daveshah1, 2020-05-19:
The problem is "Could NOT find PythonLibs". Do you have the Python3 development libraries (probably ending in -dev but may depend on the distro) installed?

Martoni, 2020-05-19:
Thanks daveshah1, that was it.
$ sudo apt install python3-dev

Sorry for this stupid question.

closed 2020-05-19 by Martoni