YosysHQ/prjtrellis #149: Kindly list the ubuntu packages to build this

tommythorn, 2020-10-05:
I don't really use Python so the description in the README isn't sufficient to enabling building this.
It wasn't obvious what I needed and I only succeeded with
```
sudo apt install python3-dev boost-python3 libboost-python-dev libboost-all-dev
```
which seems like a large overkill and doesn't cover cmake and other necessities.  I'll try to paid it down to what's needed on a fresh Ubuntu installation, but someone probably already has a better answer to the boost
question.

eine, 2020-12-28:
FTR, https://github.com/ghdl/docker/issues/24.

tommythorn, 2021-05-16:
Moved to Ubuntu 21.04/Arm64 and the above no longer works.  The following seemed to do the job but is surely incomplete:
```
sudo apt install -y git cmake libboost-all-dev qt5-quick-demos libeigen3-dev libboost-python-dev libboost-all-dev
```