YosysHQ/prjtrellis #155: tilegrid

arblake, 2020-12-05:
Does prtrellis  create and install tilegrid.json  in database ?  If not what does?

daveshah1, 2020-12-05:
tilegrid.json is obtained from Diamond via the [extract_tilegrid.py](https://github.com/YosysHQ/prjtrellis/blob/master/tools/extract_tilegrid.py) script and stored in the [prjtrellis-db repo](https://github.com/YosysHQ/prjtrellis-db/blob/master/ECP5/LFE5U-25F/tilegrid.json) as with the rest of the database - this should become this database submodule when you check out prjtrellis, and then be installed to `/usr/local/share/trellis` (or wherever else you specify) in the libtrellis CMake install script.

arblake, 2020-12-05:
Thanks   now solved

closed 2020-12-05 by arblake