YosysHQ/prjtrellis #158: Missing PINs on CABGA554 package

zguig52, 2021-01-21:
Hello,

When discovering litex with ECPIX-5 board, I was improving the board description and found some missing PINs:
ERROR: IO pin 'spiflash4x_clk$tr_io' constrained to pin 'AE3', which does not exist for package 'CABGA554'

I contacted the team in charge of the board and checked with them if it was an issue with their doc or with treillis DB. After some checks, it appears that this pin exists in the package CABGA554 as can be confirmed by lattice doc also:
https://www.latticesemi.com/view_document?document_id=51579

I will work on cross-checking this file with ECP5/LFE5UM5G-85F/iodb.json file and update the later to reflect all the pinout of this model.

Do you have kind of auto tools / scripts to extract infos like this from CSV file or shall I do this manually?

daveshah1, 2021-01-21:
You need to use the USRMCLK primitive for this pin (at least at the moment)

zguig52, 2021-01-25:
Hi daveshah1,
Thanks for your reply, this seems to make the trick. Placement works fine with this primitive, still have to check if Flash works with it, but it looks good!
Do you have somewhere in the project a list with all differences between manufacturer pins names and treillis names?

zguig52, 2021-01-25:
I see this thread which seems related:
https://github.com/YosysHQ/nextpnr/issues/290

smunaut, 2021-01-25:
Pin 'AE3' is not a FPGA IO. You can't use it as a normal pin.

There is a way to route a signal to it by instantiating a special primitive in the code : 

```
USRMCLK clk_I (
    .USRMCLKI(spi_clk_o),
    .USRMCLKTS(spo_clk_oe_n)
);
```

The spi clock will not appear at all on your top level ... and will not be in the PCF at all.

This is nothing to do with Treillis, it's just that this is not a normal FPGA pin, it has a dedicated function. You can't pack IO register, or use it as input or a bunch of other stuff ...

zguig52, 2021-01-25:
Hi smunaut,
Thanks for your explanations, I understand (a bit ;) better what is going on with this pin!
I will continue to discuss about this with litex/migen guys to check if I do the right thing related to this (it seems that with not putting the clock pin with the board spi flash block do the trick as I saw this line in the build logs:       USRMCLK                         1).

closed 2021-01-25 by zguig52