YosysHQ/prjtrellis #160: PLL generated with --highres option results in multiply driven net error

tomverbeure, 2021-01-28:
I created a PLL like this:

```
ecppll -n pll --clkin_name inclk0 --clkout0_name c0 -i 25 -o 12.288 --highres -f pll.v
```

It's instantiated in the design like this:

```verilog
    pll u_pll(.inclk0(osc_clk_in), .c0(main_clk));
```

I get the following error when using nextpnr-ecp5:

```
3.51. Executing JSON backend.

Warnings: 1 unique messages, 2 total
End of script. Logfile hash: 49c6101b52, CPU: user 0.70s system 0.08s, MEM: 264.66 MB peak
Yosys 0.9+3845 (git sha1 4762cc06c, clang 6.0.0-1ubuntu2 -fPIC -Os)
Time spent: 73% 15x read_verilog (0 sec), 8% 8x techmap (0 sec), ...
nextpnr-ecp5 --25k --package CABGA381 --speed 6 --json pll_test.json --textcfg pll_test_out.config --lpf pll_test.lpf --freq 65
ERROR: Net 'main_clk' is multiply driven by cell ports u_pll.pll_i.CLKOP and u_pll.pll_i.CLKOS
0 warnings, 1 error
```

When I create the PLL without the `--highres` option, the design compiles fine.

To reproduce: unzip the following file:

[ecp5_pll_test.zip](https://github.com/YosysHQ/nextpnr/files/5839852/ecp5_pll_test.zip)

and do `make clean && make`.

Tom

ikle, 2021-05-09:
This is the regression introduced by commit b2884bff: CLKOP and CLKFB parameters should not be named as clkout0_name.

uis246, 2021-12-19:
Close issue then?

ikle, 2022-06-10:
Let's try to reproduce the error with fixed ecppll:
```bash
ecp5_pll_test$ diff -Naurd Makefile.orig Makefile
--- Makefile.orig       2022-06-10 05:22:53.574169255 +0300
+++ Makefile    2022-06-10 05:23:16.622482326 +0300
@@ -3,9 +3,6 @@

 OBJS+=top.v pll.v

-#TRELLIS=/usr/local/share/trellis
-TRELLIS=/opt/prjtrellis/share/trellis
-
 all: ${TARGET}.bit

 $(TARGET).json: $(OBJS)
@@ -18,7 +15,7 @@
        ecppack --svf ${TARGET}.svf $< $@

 pll.v: Makefile
-       $(TRELLIS)/../../bin/ecppll -n pll --clkin_name inclk0 --clkout0_name c0 -i 25 -o 12.288 --highres -f pll.v
+       ecppll -n pll --clkin_name inclk0 --clkout0_name c0 -i 25 -o 12.288 --highres -f pll.v

 ${TARGET}.svf : ${TARGET}.bit   

ecp5_pll_test$ rm pll.v
ecp5_pll_test$ make
ecppll -n pll --clkin_name inclk0 --clkout0_name c0 -i 25 -o 12.288 --highres -f pll.v
Pll parameters:
Refclk divisor: 1
Feedback divisor: 1
clkout0 divisor: 29
clkout0 frequency: 12.2881 MHz   
clkout1 divisor: 59
clkout1 frequency: 12.2881 MHz   
clkout1 phase shift: 4.58155e-41 degrees
VCO frequency: 725
yosys -p "synth_ecp5 -json pll_test.json" top.v pll.v
...
3.51. Executing JSON backend.

End of script. Logfile hash: de1175be07, CPU: user 0.76s system 0.08s, MEM: 262.70 MB peak
Yosys 0.9+4052 (git sha1 5c1e6a0e, gcc 6.3.0-18+deb9u1 -fPIC -Os)
Time spent: 75% 15x read_verilog (0 sec), 6% 15x opt_clean (0 sec), ...
nextpnr-ecp5 --25k --package CABGA381 --speed 6 --json pll_test.json --textcfg pll_test_out.config --lpf pll_test.lpf --freq 65
Info: constraining clock net 'osc_clk_in' to 25.00 MHz

Info: Logic utilisation before packing:
Info:     Total LUT4s:        22/24288     0%
Info:         logic LUTs:      0/24288     0%
Info:         carry LUTs:     22/24288     0%
Info:           RAM LUTs:      0/12144     0%
Info:          RAMW LUTs:      0/ 6072     0%

Info:      Total DFFs:        44/24288     0%
...
Info: Program finished normally. 
ecppack --svf pll_test.svf pll_test_out.config pll_test.bit
ecp5_pll_test$ echo $?
0
ecp5_pll_test$ 
```
Looks fixed, but better @tomverbeure check it out.