YosysHQ/prjtrellis #182: versa5g example error JSON

malindralk, 2022-01-11:
I get the following error when i run "make prog"
Operating system fedora 
```

nextpnr-ecp5 --json demo.json --lpf versa.lpf --textcfg demo_out.config --um5g-45k --package CABGA381
ERROR: Failed to parse JSON file 'demo.json': unexpected end of input.
0 warnings, 1 error
make: *** [Makefile:14: demo_out.config] Error 255

```

gatecat, 2022-01-11:
Can you try `make clean` and then try again? Are any errors or warnings printed beforehand?

malindralk, 2022-01-11:
` 2.46. Executing OPT_LUT_INS pass (discard unused LUT inputs).
Optimizing LUTs in $abstract\top.

2.47. Executing AUTONAME pass.

2.48. Executing HIERARCHY pass (managing design hierarchy).

2.48.1. Executing AST frontend in derive mode using pre-parsed AST for module `\top'.
Generating RTLIL representation for module `\top'.

2.48.2. Analyzing design hierarchy..
Top module:  \top

2.48.3. Analyzing design hierarchy..
Top module:  \top
Removing unused module `$abstract\top'.
Removed 1 unused modules.

2.49. Printing statistics.

=== top ===

   Number of wires:                124
   Number of wire bits:           1662
   Number of public wires:          58
   Number of public wire bits:     784
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              4
   Number of cells:                  7
     $add                            2
     $eq                             2
     $mux                            1
     $not                            2

2.50. Executing CHECK pass (checking for obvious problems).
Checking module top...
Found and reported 0 problems.

2.51. Executing JSON backend.
ERROR: Module top contains processes, which are not supported by JSON backend (run `proc` first).
make: *** [Makefile:11: demo.json] Error 1

 `

malindralk, 2022-01-11:
The error says the " processes is not supported by JASON backend "
@gatecat

gatecat, 2022-03-31:
Hopefully this is fixed with 46f50f7

malindralk, 2022-04-01:
@gatecat  thanks

closed 2022-04-01 by malindralk