YosysHQ/prjtrellis #185: Segfault importing pytrellis.so on Apple M1

ccattuto, 2022-03-09:
OSX Monterey 12.2.1. Native arm64 Python 3.9 via Miniforge. Xcode 13.2.1. The library and executables build cleanly. pytrellis.so is very big (~7Mb). On importing it from Python, it segfaults when calling take_gil().

Tigralt, 2022-03-10:
Same issue here.

OSX Monterey 12.2.1
Native arm64
Python 3.10.2 (installed via Homebrew)
Boost 1.78.0
AppleClang 13.0.0.13000029

ccattuto, 2022-03-12:
Update: manually removing `libpython3.9.dylib` from the linker script for `pytrellis.so` fixes this.

qcabanes, 2022-03-18:
Works like a charm! I could finally install [nextpnr](https://github.com/YosysHQ/nextpnr).

ccattuto, 2022-04-08:
PS: turns out that this works out of the box if you use `gmake` instead of the (ancient) `make` packaged with OSX.

JamesTimothyMeech, 2023-04-27:
Has anyone else tried the above solution without success? Could me using python3.11 instead of 3.9 prevent the fix from working?