YosysHQ/prjtrellis #195: type_caster_base.h:517:26: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}

omcaif, 2022-09-11:
I was trying to build `libprjtrllis` with Python (i.e., with the default CMake option `BUILD_PYTHON:BOOL=ON`) for use by `nextpnr-ecp5`; however, I am getting the error shown in the title of this issue. This issue does not occur when `BUILD_PYTHON` is set to `OFF`.

Here are the full "logs" on latest commit, 98e0ea3064a22b3c9159fb4df35e1792d87a0c9e:

<details>
<summary>Console log</summary>

```
$ git clone --recursive https://github.com/YosysHQ/prjtrellis
$ cd prjtrellis/
$ source environment.sh
$ cd libtrellis/
$ cmake .
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Python3: /usr/bin/python3.11 (found suitable version "3.11.0", minimum required is "3.5") found components: Interpreter 
-- Found Python3: /usr/include/python3.11 (found suitable version "3.11.0", minimum required is "3.5") found components: Development Development.Module Development.Embed 
CMake Warning at /usr/share/cmake/Modules/FindBoost.cmake:1384 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindBoost.cmake:1507 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake/Modules/FindBoost.cmake:2118 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:59 (find_package)

CMake Warning at /usr/share/cmake/Modules/FindBoost.cmake:1384 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindBoost.cmake:1507 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake/Modules/FindBoost.cmake:2118 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:59 (find_package)

CMake Warning at /usr/share/cmake/Modules/FindBoost.cmake:1384 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindBoost.cmake:1507 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake/Modules/FindBoost.cmake:2118 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:59 (find_package)

CMake Warning at /usr/share/cmake/Modules/FindBoost.cmake:1384 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindBoost.cmake:1507 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake/Modules/FindBoost.cmake:2118 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:59 (find_package)

CMake Warning at /usr/share/cmake/Modules/FindBoost.cmake:1384 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindBoost.cmake:1507 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake/Modules/FindBoost.cmake:2118 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:59 (find_package)

CMake Warning at /usr/share/cmake/Modules/FindBoost.cmake:1384 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
[... 99 lines cut ...]
Using Clang version 15 fails with a similar error:

```
[snip]
In file included from /tmp/prjtrellis/libtrellis/src/PyTrellis.cpp:17:
In file included from /tmp/prjtrellis/libtrellis/3rdparty/pybind11/include/pybind11/pybind11.h:13:
In file included from /tmp/prjtrellis/libtrellis/3rdparty/pybind11/include/pybind11/detail/class.h:12:
In file included from /tmp/prjtrellis/libtrellis/3rdparty/pybind11/include/pybind11/detail/../attr.h:13:
In file included from /tmp/prjtrellis/libtrellis/3rdparty/pybind11/include/pybind11/cast.h:15:
/tmp/prjtrellis/libtrellis/3rdparty/pybind11/include/pybind11/detail/type_caster_base.h:517:26: error: member access into incomplete type 'PyFrameObject' (aka '_frame')
            frame = frame->f_back;
                         ^
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of '_frame'
typedef struct _frame PyFrameObject;
               ^
[snip]
```

7a2e9ed8a7d145251f5720a43229ee77de637d4a builds

7db348cbd8bdd625017e6c83a26a382c5bd6d9a6 builds

8f182f1b6d87c436b78a45d9588fbc2138e1ddc0 builds

b76481ee32ac4bc27500c7269796450783162434 does not build

gatecat, 2022-09-13:
Thanks for reporting, can you see if https://github.com/YosysHQ/prjtrellis/pull/196 fixes the issue?

Also, out of pure curiosity, what version `python3 -V` states?

omcaif, 2022-09-13:
> Thanks for reporting, can you see if #196 fixes the issue?

Yes, it does, but it still picks Python 3.11, which is kind of unexpected(?) Maybe this is a CMake issue?

> Also, out of pure curiosity, what version `python3 -V` states?

```
$ python3 --version
Python 3.10.7
$ python3.11 --version
Python 3.11.0rc1
$ cat /etc/python-exec/python-exec.conf
#/etc/python-exec/python-exec.conf
# python-exec global preference configuration
#
# This file specifies global Python preference for python-exec. It
# determines which implementations will be used when wrapped Python
# scripts or generic Python wrappers (such as /usr/bin/python) are
# started with no implementation override (EPYTHON).
#
# The global configuration can be overriden for each wrapped script
# in /etc/python-exec/<script-name>.conf. If this override exists,
# the global configuration is not used at all.
#
# The preference file should list Python implementations, one per line,
# in EPYTHON-format (e.g. python3.5), in order of preference. The most
# preferred implementation should be listed first. The program will use
# the most preferred implementation that is supported by it.
#
# If none of the listed implementations are supported, python-exec will
# attempt to use the remaining supported implementations in default
# order.
#
# A specific implementation can be disabled by preceding it with
# a hyphen (-). In this case, the implementation will never be used.
# If no other implementation is supported by the program, it will fail
# to start.
#
# Note: this file can be modified by calls to 'eselect python'.
python3.10
```

gatecat, 2022-09-16:
> Yes, it does, but it still picks Python 3.11, which is kind of unexpected(?) Maybe this is a CMake issue?

Yes, it does seem like CMake is picking the newest rather than default version

> EDIT: Should I open a separate issue for nextpnr?

the pybind stuff for 3.11 should now be sorted for nextpnr too, but if you want to create an issue for the CMake stuff that can be done too

omcaif, 2022-09-16:
> > Yes, it does, but it still picks Python 3.11, which is kind of unexpected(?) Maybe this is a CMake issue?
> 
> Yes, it does seem like CMake is picking the newest rather than default version

After some searching, it is possible to make it check the "unversioned" Python first by setting `Python3_FIND_UNVERSIONED_NAMES` to `FIRST` according to <https://gitlab.kitware.com/cmake/cmake/-/blob/cabba23ab270eb5cc672c43265f9eb3cb7dda8cc/Modules/FindPython3.cmake#L388>. I don't know if they intended to make the behavior different compared to the "deprecated" method (which was replaced by commit b76481ee32ac4bc27500c7269796450783162434 in this repo) but I do not care enough to make an account and open an issue.

In my opinion, that option should be set, but if it won't be set in this project, I can just override it myself locally in order to use an actual release of Python (instead of the non-default 3.11 release candidate).

Anyway, closing this since the original issue is solved.

closed 2022-09-16 by omcaif