YosysHQ/prjtrellis #202: ecp5-versa5g.cfg doesn't work for newer boards

hammdist, 2022-12-18:
I have the newer LFE5UM-45F-VERSA-EVNG Rev C board (2022).

I have set the jumper thing like on an older board (2015) that is able to flash using openocd/prjtrellis.

It doesn't work.

Here is the console output from attempting to program the board: https://paste.ee/p/FOy1v

I have tried to change the -expected-id 0x81112043 in the script to the id that is found, but that doesn't make the tdo error go away.

gatecat, 2022-12-18:
Sounds like your board has one of the parts without 5G serdes. Also replace --um5g-45k with --um-45k in the makefile and rebuild the bitstream.

hammdist, 2022-12-18:
Indeed the marking on the fpga part indicates that it is not 5G, even though the silkscreen on the board says it is a 5G versa development kit.

I have tried replacing --um-45k on my project but I get another error ... when I get a chance I will try it on the demo project to rule out my project as the cause of the error, and post the complete error.

hammdist, 2022-12-18:
Yeah, I still get the flashing error even with the demo.

https://paste.ee/p/GspQN

hammdist, 2022-12-18:
The error seems to occur at the end of the `svf` file:

```
 25887	SIR	8	TDI  (FF);
 25888	RUNTEST	IDLE	100 TCK	1.00E-02 SEC;
 25889	
 25890	SIR	8	TDI  (C0);
 25891	RUNTEST	IDLE	2 TCK	1.00E-03 SEC;
 25892	SDR	32	TDI  (00000000)
 25893				TDO  (00000000)
 25894				MASK (FFFFFFFF);
 25895	
 25896	SIR	8	TDI  (26);
 25897	RUNTEST	IDLE	2 TCK	2.00E-01 SEC;
 25898	
 25899	SIR	8	TDI  (FF);
 25900	RUNTEST	IDLE	2 TCK	1.00E-03 SEC;
 25901	
 25902	SIR	8	TDI  (3C);
 25903	SDR	32	TDI  (00000000)
 25904				TDO  (00000100)
 25905				MASK (00002100);
```

gatecat, 2022-12-18:
Oh, did you try the jumper instructions at https://github.com/YosysHQ/prjtrellis/tree/master/examples/versa5g too?

hammdist, 2022-12-18:
Yeah, the jumpers are set correctly (pretty sure about that).

hammdist, 2022-12-22:
Oops, took another look at this and discovered that I had probably forgot to `make clean` after updating the `Makefile` to specify `--um-45k`. After rebuilding properly, it is sort of working now in that I get counter output on the LEDs but (strangely) nothing on the segment display.

closed 2022-12-22 by hammdist

gparmeggiani, 2023-12-18:
Hello @hammdist 
Did you perhaps solve the issue with the segment display?

I also can see the LEDs do the binary count, but the alphanumeric display is completely off.