YosysHQ/prjtrellis #216: Toolchain build failure

keithdennison, 2023-03-25:
A previously working yosys/trellis/nextpnr (ecp5) toolchain build is now failing. The trailing output is:

ecppack --svf testpins.svf testpins_out.config testpins.bit
terminate called after throwing an instance of 'boost::wrapexcept<boost::property_tree::ptree_bad_path>'
  what():  No such node (row_bias)

I have narrowed it down to the commit 9e9fd8c where it first appears.

mmicko, 2023-03-25:
Please do check if if submodule is updated correctly at your git tree. Using `git submodule` should return:
```
db49a4a62bc945ad30a5aedf2f4c4d2e3814a661 database (heads/master)
```
If not use next line to update it (you may need to git reset --hard or similar in case you have local changes) :
```
git submodule update
```

In case this all is fine, it could be that your installation is using latest binaries but old database, since database.json should contain row_bias for all devices as in current git.

keithdennison, 2023-03-27:
Thank you. It worked after a "git submodule update" (and subsequent make/install).

closed 2023-03-27 by keithdennison