YosysHQ/prjtrellis #217: Build of nextpnr fails with newest prjtrellis GIT version

janschiefer, 2023-03-26:
The build for nextpnr fails for the lastest prjtrellis git:

Prjtrellis: b9120de2453e71b68b344452ef12040ee53d47ab
Nextpnr: b36e8a3013ac70a9fbe71d2163f660dafe3b8b2f

```
pip3 install apycula --upgrade
make clean
cmake -DARCH="ice40;ecp5;gowin" -DTRELLIS_INSTALL_PREFIX=/usr/local -DGOWIN_BBA_EXECUTABLE=/home/jschiefer/.local/bin/gowin_bba .
make -j32

Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: apycula in /home/jschiefer/.local/lib/python3.10/site-packages (0.7)
Requirement already satisfied: crcmod in /home/jschiefer/.local/lib/python3.10/site-packages (from apycula) (1.7)
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from apycula) (1.21.5)
-- Building with IPO
-- Found Python3: /usr/include/python3.10 (found suitable version "3.10.7", minimum required is "3.5") found components: Development Development.Module Development.Embed 
-- Found Boost: /usr/include (found version "1.74.0") found components: filesystem program_options iostreams system thread regex chrono date_time atomic 
-- Found Boost: /usr/include (found version "1.74.0") found components: program_options filesystem system 
-- Configuring architecture: ice40
-- Enabled iCE40 devices: 384;1k;5k;u4k;8k
-- Found Python3: /usr/bin/python3.10 (found suitable version "3.10.7", minimum required is "3.5") found components: Interpreter 
-- IceStorm install prefix: /usr/local
-- icebox data directory: /usr/local/share/icebox
-- Using iCE40 chipdb: /home/jschiefer/Code/nextpnr/ice40/chipdb
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Trellis install prefix: /usr/local
-- Trellis library directory: /usr/local/lib/trellis
-- Trellis data directory: /usr/local/share/trellis
-- Using ECP5 chipdb: /home/jschiefer/Code/nextpnr/ecp5/chipdb
-- Configuring architecture: gowin
-- Enabled Gowin devices: GW1N-1;GW1NZ-1;GW1N-4;GW1N-9;GW1N-9C;GW1NS-2;GW1NS-4
-- gowin_bba executable: /home/jschiefer/.local/bin/gowin_bba
-- Using Gowin chipdb: /home/jschiefer/Code/nextpnr/gowin/chipdb
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jschiefer/Code/nextpnr
[  0%] Generating chipdb/chipdb-384.bba
[  0%] Building CXX object bba/CMakeFiles/bbasm.dir/main.cc.o
[  1%] Generating chipdb/chipdb-1k.bba
[  2%] Linking CXX executable bbasm
[  2%] Built target bbasm
[  3%] Generating chipdb/chipdb-5k.bba
[  4%] Generating chipdb/chipdb-u4k.bba
[  4%] Generating chipdb/chipdb-8k.bba
[  4%] Built target chipdb-ice40-bbas
[  4%] Generating chipdb/chipdb-25k.bba
[  5%] Generating ice40/chipdb/chipdb-1k.cc
[  5%] Generating ice40/chipdb/chipdb-384.cc
[  5%] Generating ice40/chipdb/chipdb-5k.cc
[  5%] Generating ice40/chipdb/chipdb-u4k.cc
[  6%] Generating ice40/chipdb/chipdb-8k.cc
Traceback (most recent call last):
  File "/home/jschiefer/Code/nextpnr/ecp5/trellis_import.py", line 701, in <module>
    main()
  File "/home/jschiefer/Code/nextpnr/ecp5/trellis_import.py", line 687, in main
    chip = pytrellis.Chip(dev_names[args.device])
RuntimeError: No such node (row_bias)
make[2]: *** [ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/build.make:77: ecp5/chipdb/chipdb-25k.bba] Error 1
make[1]: *** [CMakeFiles/Makefile2:489: ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  6%] Built target chipdb-ice40-bins
make: *** [Makefile:136: all] Error 2

```

janschiefer, 2023-03-26:
The breaking change was introduced in prjtrellis with this commit: 9e9fd8c65463094f71e24a3b96056514d861625d

Building with an earlier version works fine.

whitequark, 2023-03-26:
That's interesting--the YoWASP build succeeded: https://github.com/YoWASP/nextpnr/actions/runs/4506793983/jobs/7933952077

mmicko, 2023-03-26:
Please do first compare device.json in prjtrellis root and one in database directory. If those two are not the same that points to a problem of database submodule not updated.
If that is fine check if device.json from install location is same, if not you just need to run make install again.

To fix submodule issue do next.
Using git submodule should return:

`db49a4a62bc945ad30a5aedf2f4c4d2e3814a661 database (heads/master)`

If not use next line to update it (you may need to git reset --hard or similar in case you have local changes) 
`git submodule update`

Also make sure to do make install from libtrellis directory, to copy latest database to /usr/local

janschiefer, 2023-03-27:
git submodule:

-35d900a94ff0db152679a67bf6e4fbf40ebc34aa database

git reset --hard
git submodule update
git submodule 

-35d900a94ff0db152679a67bf6e4fbf40ebc34aa database

janschiefer, 2023-03-27:
Instresting. 
After launching the commands from above the build works.

However if I do...

git pull --rebase origin master

and 

git reset --hard
git submodule update
( git submodule outputs " -db49a4a62bc945ad30a5aedf2f4c4d2e3814a661 database " here )

I get the same errors as above...

mmicko, 2023-03-28:
Well `git status` would give you if you are actually on some other branch currently.
I looks that you are, since rebase have helped.

janschiefer, 2023-03-29:
Working state reverted to before https://github.com/YosysHQ/prjtrellis/commit/9e9fd8c65463094f71e24a3b96056514d861625d
(update.sh is just my git update script with " git pull --rebase origin master " in it)

On branch master
Your branch is behind 'origin/master' by 37 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	update.sh

nothing added to commit but untracked files present (use "git add" to track)

git status on not working current git version:
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	update.sh

nothing added to commit but untracked files present (use "git add" to track)

janschiefer, 2023-03-30:
The problem diasspeared after deleting all local files and a new "git checkout --recursive" of prjtrellis and nextpnr.

Seems like a git misconfiguration isse on my side.

closed 2023-03-30 by janschiefer