YosysHQ/prjtrellis #222: ecppack crashes with "No such node (row_bias)"

kittennbfive, 2023-05-13:
Hi,

sorry to bother you, but it looks like ecppack is broken...

**Versions**
```
$ yosys --version
Yosys 0.28+12 (git sha1 4251d37f4, clang 11.0.1-2 -fPIC -Os)
$ nextpnr-ecp5 --version
"nextpnr-ecp5" -- Next Generation Place and Route (Version nextpnr-0.5-159-g91771895)
$ ecppack --version
Project Trellis ecppack Version 1.3-145-gc99b22d
```
(compiled from current git master a few days ago / just a sec ago for prjtrellis)

**Workflow**
With the example files (CC0) attached:
[files.zip](https://github.com/YosysHQ/prjtrellis/files/11469972/files.zip)
```
yosys -p "synth_ecp5 -json out.json -top test_inputs" test_inp.v && nextpnr-ecp5 --json out.json --25k --package CABGA256 --lpf constraints.lpf --textcfg out.cfg -q && ecppack --svf bitstream.svf out.cfg
```
**Problem**
`ecppack` crashes: 
```
terminate called after throwing an instance of 'boost::wrapexcept<boost::property_tree::ptree_bad_path>'
  what():  No such node (row_bias)
```
I am afraid i don't know C++ so i won't be able to debug this myself. Thanks for your help and the great work.

kittennbfive, 2023-05-13:
gdb backtrace if it can help:
```
Starting program: /usr/bin/ecppack --svf bitstream.svf out.cfg
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
terminate called after throwing an instance of 'boost::wrapexcept<boost::property_tree::ptree_bad_path>'
  what():  No such node (row_bias)

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) backtrace
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7a64537 in __GI_abort () at abort.c:79
#2  0x00007ffff7ccc7ec in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff7cd7966 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff7cd79d1 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff7cd7c65 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff7f6fd4d in boost::property_tree::basic_ptree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::get_child(boost::property_tree::string_path<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, boost::property_tree::id_translator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) () from /usr/bin/../lib/x86_64-linux-gnu/trellis/libtrellis.so
#7  0x00007ffff7f69155 in Trellis::get_chip_info(Trellis::DeviceLocator const&) () from /usr/bin/../lib/x86_64-linux-gnu/trellis/libtrellis.so
#8  0x00007ffff7f56a8f in Trellis::Chip::Chip(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
   from /usr/bin/../lib/x86_64-linux-gnu/trellis/libtrellis.so
#9  0x00007ffff7f5b761 in Trellis::ChipConfig::to_chip() const () from /usr/bin/../lib/x86_64-linux-gnu/trellis/libtrellis.so
#10 0x000055555555f4d8 in main ()
```

mmicko, 2023-05-13:
Looks like your "database" submodule is not properly updated, check https://github.com/YosysHQ/prjtrellis/issues/217 how to do it, and then make sure to "make install" again

kittennbfive, 2023-05-13:
Thank you! Indeed i didn't use the correct git commands, how embarassing... I will put a note on my github how to update the toolchain if this can help anybody.

closed 2023-05-13 by kittennbfive