YosysHQ/prjtrellis #225: Please clarify licensing of timing/resource/nescore.v

DanielG, 2023-06-13:
Hi,

I'm working on packaging project trellis for Debian. One of the first steps of doing this is copyright review. I noticed that `timing/resource/nescore.v` includes a notice stating "This program is GPL Licensed. See COPYING for the full license."

Just "GPL" isn't really very specific it could be any of GPL-{1,2,3}.0-{only,or-later}. However the mentioned COPYING file is nowhere to be found. Now obviously this was copied over from some other project, https://github.com/daveshah1/up5k-demos/tree/master/nes is mentioned at the top so that would be my guess. However that doesn't contain any more specific licensing information either.

Thanks,
--Daniel

DanielG, 2023-08-25:
I just had prjtrellis REJECTed by Debian ftp-master because of this.

@gatecat can you shed some light on this perhaps?

gatecat, 2023-08-25:
I'll look into a proper headed, but this file isn't needed at runtime, nor does any content make its way into any file that does in any way that would meaningfully affect copyright or licensing (it's just a test design run through Diamond, with a net and cell delay model extracted from the design that Diamond builds).

DanielG, 2023-08-25:
It does affect licensing insofar as Debian would be distributing it as part of the source tarball, unfortunately ftp-master is particularly anal about the mere redistribution aspect of copyright.

I just find it disappointing if it's not possible the re-run the fuzzing process using the source tarball from Debian, but you're right ofc. it ultimately doesn't matter and I'm just excluding it from our tarball until this is resolved.

--Daniel