YosysHQ/prjtrellis #239: Add support for TG144 package

machdyne, 2024-02-07:
Hello,

I've designed a board (["Vanille"](https://github.com/machdyne/vanille)) using the LFE5U-12F-6TG144C. There doesn't appear to be any support for the TG144 (144-pin TQFP) package in Project Trellis at the moment. I would like to help to add support for this package, but I'm not quite sure how to get started. Any advice would be appreciated.

gatecat, 2024-02-07:
This should already be supported as of https://github.com/YosysHQ/prjtrellis-db/commit/0110372c4b77223ab14c65bfb2876c68f5fc0845 and https://github.com/YosysHQ/nextpnr/commit/88714c54ec9cf556af0da7c6b5cb6ad0e9e30032.

If you're having any problems with it, let me know.

machdyne, 2024-02-07:
That's great news. I guess I was looking in the wrong place. Thanks!

machdyne, 2024-02-29:
FWIW, the board has now been tested and everything seems to be working fine with the TQFP144 package.

closed 2024-02-29 by machdyne