YosysHQ/prjtrellis #242: ecppll random (uninitialised?) values in output

BrettRD, 2024-11-08:
Running ecppll twice with the same input returns different results

```
$ ecppll -i 60 -o 120 --highres -f first.v
$ ecppll -i 60 -o 120 --highres -f second.v
$ diff first.v second.v 
32,33c32,33
<         .CLKOS_CPHASE(-895312416),
<         .CLKOS_FPHASE(32676),
---
>         .CLKOS_CPHASE(-1799184928),
>         .CLKOS_FPHASE(32738),
```

Specifying three output frequencies has the same random behaviour

Dropping the `--high-res` flag makes the output stable

kittennbfive, 2025-05-04:
Indeed, i can reproduce this with `Version 1.4-69-g14ac883` (somewhat older). What version do you use?

kittennbfive, 2025-05-04:
BTW, ecppll.cpp has inconsistent formatting it looks like.