YosysHQ/prjtrellis #249: [ecppll] ecppll doesn't specify the unused PLL signals which triggers lint errors.

zapta, 2025-07-10:
The following command:
```
ecppll -i 25 -o 120 -f pll.v
```

Generates this verilog module:
```
module pll
(
    input clkin, // 25 MHz, 0 deg
    output clkout0, // 120 MHz, 0 deg
    output locked
);
(* FREQUENCY_PIN_CLKI="25" *)
(* FREQUENCY_PIN_CLKOP="120" *)
(* ICP_CURRENT="12" *) (* LPF_RESISTOR="8" *) (* MFG_ENABLE_FILTEROPAMP="1" *) (* MFG_GMCREF_SEL="2" *)
EHXPLLL #(
        .PLLRST_ENA("DISABLED"),
        .INTFB_WAKE("DISABLED"),
        .STDBY_ENABLE("DISABLED"),
        .DPHASE_SOURCE("DISABLED"),
        .OUTDIVIDER_MUXA("DIVA"),
        .OUTDIVIDER_MUXB("DIVB"),
        .OUTDIVIDER_MUXC("DIVC"),
        .OUTDIVIDER_MUXD("DIVD"),
        .CLKI_DIV(5),
        .CLKOP_ENABLE("ENABLED"),
        .CLKOP_DIV(5),
        .CLKOP_CPHASE(2),
        .CLKOP_FPHASE(0),
        .FEEDBK_PATH("CLKOP"),
        .CLKFB_DIV(24)
    ) pll_i (
        .RST(1'b0),
        .STDBY(1'b0),
        .CLKI(clkin),
        .CLKOP(clkout0),
        .CLKFB(clkout0),
        .CLKINTFB(),
        .PHASESEL0(1'b0),
        .PHASESEL1(1'b0),
        .PHASEDIR(1'b1),
        .PHASESTEP(1'b1),
        .PHASELOADREG(1'b1),
        .PLLWAKESYNC(1'b0),
        .ENCLKOP(1'b0),
        .LOCK(locked)
	);
endmodule
```

Which triggers lint errors for the missing PLL signals that are omitted from the instantiation:

```
verilator_bin --lint-only --quiet --bbox-unsup --timing -Wno-TIMESCALEMOD -Wno-MULTITOP -DAPIO_SIM=0 --top-module main -I"/Users/user/.apio/packages/oss-cad-suite/share/yosys/ecp5" _build/default/hardware.vlt "/Users/user/.apio/packages/oss-cad-suite/share/yosys/ecp5/cells_sim.v" "/Users/user/.apio/packages/oss-cad-suite/share/yosys/ecp5/cells_bb.v" main.v pll.v

%Warning-PINMISSING: pll.v:30:7: Instance has missing pin: 'ENCLKOS'
   30 |     ) pll_i (
      |       ^~~~~
                     /Users/user/.apio/packages/oss-cad-suite/share/yosys/ecp5/cells_bb.v:124:22: ... Location of port declaration
  124 |  input RST, ENCLKOP, ENCLKOS, ENCLKOS2, ENCLKOS3,
      |                      ^~~~~~~
                     ... For warning description see https://verilator.org/warn/PINMISSING?v=5.037
                     ... Use "/* verilator lint_off PINMISSING */" and lint_on around source to disable this message.
%Warning-PINMISSING: pll.v:30:7: Instance has missing pin: 'ENCLKOS2'
   30 |     ) pll_i (
      |       ^~~~~
                     /Users/user/.apio/packages/oss-cad-suite/share/yosys/ecp5/cells_bb.v:124:31: ... Location of port declaration
  124 |  input RST, ENCLKOP, ENCLKOS, ENCLKOS2, ENCLKOS3,
      |                               ^~~~~~~~
%Warning-PINMISSING: pll.v:30:7: Instance has missing pin: 'ENCLKOS3'
   30 |     ) pll_i (
      |       ^~~~~
                     /Users/user/.apio/packages/oss-cad-suite/share/yosys/ecp5/cells_bb.v:124:41: ... Location of port declaration
  124 |  input RST, ENCLKOP, ENCLKOS, ENCLKOS2, ENCLKOS3,
      |                                         ^~~~~~~~
%Warning-PINMISSING: pll.v:30:7: Instance has missing pin: 'CLKOS'
   30 |     ) pll_i (
      |       ^~~~~
                     /Users/user/.apio/packages/oss-cad-suite/share/yosys/ecp5/cells_bb.v:125:16: ... Location of port declaration
  125 |  output CLKOP, CLKOS, CLKOS2, CLKOS3,
      |                ^~~~~
%Warning-PINMISSING: pll.v:30:7: Instance has missing pin: 'CLKOS2'
   30 |     ) pll_i (
      |       ^~~~~
                     /Users/user/.apio/packages/oss-cad-suite/share/yosys/ecp5/cells_bb.v:125:23: ... Location of port declaration
  125 |  output CLKOP, CLKOS, CLKOS2, CLKOS3,
      |                       ^~~~~~
%Warning-PINMISSING: pll.v:30:7: Instance has missing pin: 'CLKOS3'
   30 |     ) pll_i (
      |       ^~~~~
                     /Users/user/.apio/packages/oss-cad-suite/share/yosys/ecp5/cells_bb.v:125:31: ... Location of port declaration
  125 |  output CLKOP, CLKOS, CLKOS2, CLKOS3,
      |                               ^~~~~~
%Warning-PINMISSING: pll.v:30:7: Instance has missing pin: 'INTLOCK'
   30 |     ) pll_i (
      |       ^~~~~
                     /Users/user/.apio/packages/oss-cad-suite/share/yosys/ecp5/cells_bb.v:126:15: ... Location of port declaration
  126 |  output LOCK, INTLOCK,
      |               ^~~~~~~
%Warning-PINMISSING: pll.v:30:7: Instance has missing pin: 'REFCLK'
   30 |     ) pll_i (
      |       ^~~~~
                     /Users/user/.apio/packages/oss-cad-suite/share/yosys/ecp5/cells_bb.v:127:9: ... Location of port declaration
  127 |  output REFCLK, CLKINTFB
      |         ^~~~~~
scons: *** [_build/default/hardware] Error 1
```

The fix is to include the unused signals in the PLL instantiation.

```
      /* Unused */
      .ENCLKOS(),
      .ENCLKOS2(),
      .ENCLKOS3(),
      .CLKOS(),
      .CLKOS2(),
      .CLKOS3(),
      .INTLOCK(),
      .REFCLK()
```

whitequark, 2026-09-20:
Fixed in https://github.com/YosysHQ/prjtrellis/pull/250

closed 2026-09-20 by whitequark

zapta, 2026-09-20:
Thanks @whitequark!